zephyrproject-rtos/zephyr #49927: Bluetooth: SMP: Pairing Failed as response to Identity Address Information

alwa-nordic, 2022-09-05:
## Abstract
There is no way for the Peripheral to fail a bonding process based on the Identity Address Information from the Central.

## Details
Assume the Central initially uses a random address and, as part of bonding, distributes its Identity Address Information. Assume the peripheral decides to reject the bond upon learning the Identity Address. Maybe it already has a bond with that address and disallows re-bonding.

The spec seems to facilitate this. The following section uses the IRK as an example, but it's reasonable to think that the same holds for the Identity Address.

> A device may reject a distributed key by sending the Pairing Failed command with the reason set to "Key Rejected". This can be done, for example, if a device receives an IRK from the remote device that it already has in its security database.
-- Vol 3 Part H 3.6.1

But this is contradicted by the following sections if the Central's Identity Address Information is the last distributed key.

> \[The Pairing Failed] command may be sent at any time during the pairing process by either device in response to a message from the remote device.".
-- Vol 3 Part H 3.5.5

> Pairing is a three-phase process. The first two phases are always used and may be followed by an optional transport specific key distribution phase (see Figure 2.1):
> • Phase 1: Pairing Feature Exchange
> • Phase 2 (LE legacy pairing): Short Term Key (STK) Generation
> • Phase 2 (LE Secure Connections): Long Term Key (LTK) Generation
> • Phase 3: Transport Specific Key Distribution
-- Vol 3 Part H 2.1

> Key distribution is complete in the device sending the final key when it receives the Baseband acknowledgment for that key and is complete in the receiving device when it receives the final key being distributed.
-- Vol 3 Part H 3.6.1

There is no SMP message following the key distribution. The Zephyr Host acting as the Central, (which sends the final key), invokes `pairing_complete` immediately upon sending the last key. This is correct per spec.

If the Peripheral subsequently refuses the bond, the Peripheral application will get `pairing_failed`, while the Central shows a "<wrn> bt_smp: Unexpected SMP code 0x05" (Pairing Failed) in the logs. It's unexpected because there is no pairing process.

In the end, there is disagreement between two devices on the security properties of the link.

## Proposals for remedies
- ### Peripheral: Disconnect if Pairing Failed is not possible
  This prevents a connection in a confused state. It does however leave a "successful" bond on the Central that is not present on the Peripheral. This is bad UX.
- ### Central: Accept a Pairing Failed outside the paring process.
  If the current connection is in a bonded context, this can be a signal to delete this bond. Essentially, this overloads Pairing Failed to also mean "I'm deleting the bond with the current LTK on my side.". This can be equivalent to calling `bt_unpair`.
- ### Central: Add a small time window on the Central before finishing the pairing process.
  The Peripheral can then issue a Pairing Failed in that window. It unclear how large this window should be.
- ### Spec: Add a "Pairing Commit" SMP command.
  Backwards compatible addition if it's guarded by a feature flag. This would frame the pairing process between a Pairing Request and  one of Pairing Failed and Pairing Commit.
- ### Send the Identity Address Information (IRK) before the Identity Information (Identity address).
  It is my view that a IRK points to one Identity address. The converse is not true. I.e. Multiple IRK can point to the same Identity address. Under this assumption, if the Identity address is not recognized, the IRK will also be not recognized. This change would definitely be not backwards compatible. The order of key distribution is completely specified, and it is different.

theob-pro, 2023-02-16:
If I read this correctly, https://github.com/zephyrproject-rtos/zephyr/pull/49634 makes it that this issue is now in fact a bug. Because if rebonding is disabled, then the other device can indeed get a false sense that it has bonded, while the zephyr app will get an error.

It seems you even [added a comment](https://github.com/zephyrproject-rtos/zephyr/pull/49634/files#diff-8dd16bb0acc8dfec9533e05e88e6d9bd021004d2ef5efdc51dcd8544ba269d7bR2006). If this is true, could you convert the issue to a bug (or create a new one)?

alwa-nordic, 2023-02-22:
Proposal:
Treat a disconnected event with `BT_HCI_ERR_AUTH_FAIL` over an encrypted link as a notice from the remote that it's deleting the bond and that we should do the same.

This should be spec compliant. (Hopefully peers don't use `BT_HCI_ERR_AUTH_FAIL` for no good reason.) This behavior could be proposed for inclusion in the Bluetooth standard.

@seko-nordic thoughts?

alwa-nordic, 2023-03-01:
@seko-nordic This is an issue concerning the user interface / user experience. The issue appears when a user deletes a bond on the central (e.g. a PC), but not on the peripheral (e.g. a Zephyr keyboard). When the user tries to use the keyboard with that same PC again, the keyboard will be happy to accept a bonding procedure. The PC will then say to the user  that the device has been bonded. But at that same moment, the keyboard notices that it already has a previous bond to the PC. The keyboard has the choice of deleting the old bond "allowing rebonding", or not. If rebonding is not allowed, the keyboard will internally not complete the bonding procedure, but instead immediately disconnect. The PC will try to reconnect to the keyboard and get a LTK missing message. The PC will delete its bond. And the whole interaction starts again. To break out of the loop, the user has to manually delete the bond on the keyboard. But this is not at all obvious since the PC seems to be successfully create the bond each time. To the user, it seems like the keyboard is malfunctioning.

alwa-nordic, 2023-03-01:
This brushes up against the question of what does it mean to re-bond? If there is some data associated with the bond, like settings, or crypto keys/secrets, or cached sensitive data, should it be deleted when the re-bonding happens? Should the API make it possible for the application to detect the re-bonding and make this decision for itself?

seko-nordic, 2023-03-16:
> ... tries to use the keyboard with that same PC again, the keyboard will be happy to accept a bonding procedure.

Should not the keyboard delete old bonding at the moment when it sees the new bonding request from the same PC? I guess PC uses the same identity when it connects to keyboard, so keyboard can detected that it's the same PC at the moment when connection is established. Then, when PC sends pairing request, keyboard should understand that bonding is lost on PC side and delete bonding locally.

alwa-nordic, 2023-03-17:
> Should not the keyboard delete old bonding at the moment when it sees the new bonding request from the same PC? I guess PC uses the same identity when it connects to keyboard, so keyboard can [...]

Good point. I think I need to re-evaluate my problem. I'll think out loud:

The bad situation is "handled" in code here:

https://github.com/zephyrproject-rtos/zephyr/blob/6648da731bb9882309457a3ab6ce710499ce93b8/subsys/bluetooth/host/smp.c#L1746-L1757

This code is triggered in system test `/zephyr/tests/bluetooth/bsim/host/security/bond_overwrite_denied/`.

https://github.com/zephyrproject-rtos/zephyr/blob/2a9eda226ba4747b3016aae896c47fe692bf87ff/tests/bluetooth/bsim/host/security/bond_overwrite_denied/src/peripheral.c#L30-L43

But what you said made me think that the peripheral should notice it already has a bond to the same remote identity / IRK on local id 1 and either..

 - ..refuse to respond to a connection from central to peripheral id 2.
 - ..fail the bonding from central to peripheral id 2 early.