zephyrproject-rtos/zephyr #49938: drivers/modem/gsm_ppp.c:  unnecessary modem_cmd_handler_tx_lock when CONFIG_GSM_MUX disabled

PierrickGuitter, 2022-09-05:
**Describe the bug**
Trying to lock a semaphore through modem_cmd_handler_tx_lock  methods, that is never unlocked when CONFIG_GSM_MUX is disabled.
The locking attempt times out after GSM_CMD_LOCK_TIMEOUT seconds.

 - What target platform are you using?
 Custom target with STM32U585 micro and Quectel BG950 gsm modem.
 - What have you tried to diagnose or workaround this issue?
 Used Debugger to diagnose. Add CONFIG_GSM_MUX  to protect call on modem_cmd_handler_tx_lock.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. west build -b custom_board . -p
3. west flash
4. See error

**Expected behavior**
No locking attempt performed when CONFIG_GSM_MUX is disabled.

**Impact**

**Logs and console output**
none

**Environment (please complete the following information):**
 - OS: wsl
 - Zephyr SDK
 - Seen on version zephyr-v3.0.0-1091-gca84f0dcbbed

**Additional context**
A call to modem_cmd_handler_tx_lock must be preceeded by a call of modem_cmd_handler_tx_unlock.
As modem_cmd_handler_tx_unlock is only alled under certain circumstances (CONFIG_GSM_MUX enabled), modem_cmd_handler_tx_lock should also be performed when CONFIG_GSM_MUX  is enabled.

jukkar, 2022-09-06:
@PierrickGuitter Thanks for the analysis, would you be able to send a patch that fixes the issue?

PierrickGuitter, 2022-09-06:
@jukkar Here is the patch asked for.
If there is any question about it, please tell me.
[fix.zip](https://github.com/zephyrproject-rtos/zephyr/files/9496515/fix.zip)

jukkar, 2022-09-06:
Please submit a PR as then that can be merged after a review.

fabiobaltieri, 2022-09-06:
Hey @PierrickGuitter, if you need help creating a pull request, this is the documentation for the contribution guideline and workflow: https://docs.zephyrproject.org/latest/contribute/guidelines.html#contribution-workflow.

closed 2022-09-27 by fabiobaltieri