zephyrproject-rtos/zephyr #49939: stm32 adc driver_api test fails on stm32wb55 and stm32l5

FRASTM, 2022-09-05:
**Describe the bug**
tests/drivers/adc/adc_api test fails when running on 
-     nucleo_wb55rg
-    stm32l562e_dk

Test execution blocking in the first sequence _test_adc_asynchronous_call_

Please also mention any information which could help others to understand
the problem you're facing:
 - nucleo_wb55rg board or stm32l562e_dk disco kit
- Looks like a regression after https://github.com/zephyrproject-rtos/zephyr/pull/49039 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_wb55rg tests/drivers/adc/adc_api/3. make
4. west flash

**Expected behavior**                                                                                    
```
------ TESTSUITE SUMMARY START ------                                                
                                                                                     
SUITE PASS - 100.00% [adc_basic]: pass = 5, fail = 0, skip = 1, total = 6 duration =s
 - PASS - [adc_basic.test_adc_asynchronous_call] duration = 0.007 seconds            
 - PASS - [adc_basic.test_adc_invalid_request] duration = 0.012 seconds              
 - PASS - [adc_basic.test_adc_repeated_samplings] duration = 0.098 seconds           
 - PASS - [adc_basic.test_adc_sample_one_channel] duration = 0.008 seconds           
 - SKIP - [adc_basic.test_adc_sample_two_channels] duration = 0.001 seconds          
 - PASS - [adc_basic.test_adc_sample_with_interval] duration = 0.410 seconds         
                                                                                     
------ TESTSUITE SUMMARY END ------                                                  
                                                                                     
===================================================================                  
PROJECT EXECUTION SUCCESSFUL   
```

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.1.0-4714-ge5b049249eab  ***                    
Running TESTSUITE adc_basic                                                          
===================================================================                  
START - test_adc_asynchronous_call

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2022-09-05:
Could be a mistmatch between channel and index in the adc_stm32_teardown_channels() function call  : 
Suggested correction:

```
diff --git a/drivers/adc/adc_stm32.c b/drivers/adc/adc_stm32.c
index 2015601b2a..69541a392c 100644
--- a/drivers/adc/adc_stm32.c
+++ b/drivers/adc/adc_stm32.c
@@ -901,7 +901,7 @@ static int start_read(const struct device *dev,
 
 	err = adc_context_wait_for_completion(&data->ctx);
 
-	adc_stm32_teardown_channels(dev, index);
+	adc_stm32_teardown_channels(dev, channel);
 
 	return err;
 }
```

ycsin, 2022-09-05:
Not sure what could be the cause yet, but currently the [`adc_stm32_teardown_channels`](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/adc/adc_stm32.c#L904) is getting the same arg as the [`adc_stm32_setup_channels`](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/adc/adc_stm32.c#L717) since two of them are similar

FRASTM, 2022-09-13:
On both stm32wb55 and stm32L5x , the VREFINT channel is on the **channel 0** of the ADC1/2
During the adc_stm32_teardown_channels 

One possible WA for that test execution is to remove the `has-vref-channel` property (with boards/overlay file) 
```
&adc1 {
	/delete-property/ has-vref-channel;
};
```

FRASTM, 2022-09-13:
However, the adc_stm32_teardown_channels when the channel ID is 0 --> VREFINT is  not enabling  correctly this channel but because it waits endlessly for LL_ADC_IsActiveFlag_ADRDY.

closed 2022-09-14 by fabiobaltieri