zephyrproject-rtos/zephyr #49940: Allow extending upstream boards in applications

gmarull, 2022-09-05:
**Is your feature request related to a problem? Please describe.**

I found myself in the need to extend an upstream board in an application. By extending, in this case, I mean adding new hardware description to the upstream board DTS. The reason such hardware description is not upstream is that the application implements new driver classes for specific hardware IP in the platform (STM32 Timers, which are a kind of multi-function device). So, this is something hard/not possible to upstream.

```
&timers1 {
	svpwm: svpwm {
		/* uses STM32 timer to create SV-PWM modulated signals */
		compatible = "st,stm32-svpwm";
		pinctrl-0 = <&tim1_ch1_pa8 &tim1_ch2_pa9 &tim1_ch3_pa10 &tim1_ocp_pa11>;
		pinctrl-names = "default";
		...
	};
};
```

The application module also provides shields that depend on `svpwm`, and this is where the problem starts. There is no way to extend a board before the shield overlays are applied. Which means that I can't rely on the Zephyr shield facilities.

There are other valid usecases:

- Be able to extend (for all apps) `chosen` nodes with application-specific subsystem choices (see NCS)
- Be able to override (for all apps) an upstream board choice for all applications/samples

**Describe the solution you'd like**

A simple mechanism to provide board extensions. For example, if I have `boards/arch/upstreamboard/...` in my application, with for example an overlay to have the final DTS result in:

- Upstream board.dts
- My application board.overlay
- Shield overlays
- Application overlays

**Describe alternatives you've considered**

- Be able to control shield overlay inclusion order and rely on application/shield specific overlays
- Do not use shields but a custom mechanism to ensure proper inclusion order

**Additional context**

N/A

nordicjm, 2022-09-06:
Would also be useful for building mcuboot with custom configurations without having to add an overlay directly to the mcuboot repository

JordanYates, 2022-09-25:
I suspect #40669 would be a way of solving this issue (might require additional search paths for snippets)

PR #57390 (2023-08-21): cmake: modules: add support for board extensions
Board extensions are board fragments that may be present in any board root folder, under `${BOARD_ROOT}/extensions/${BOARD}`. The board extension directory may contain Kconfig fragments and/or devicetree overlays.  Board extensions are automatically loaded and applied on top of board files, before anything else. There is no guarantee on which order extensions are applied, in case multiple exist.

Board extensions may be useful in the context of Zephyr modules, such as `example-application`. In some situations, certain hardware description or cho [... cut]
closed 2023-08-21 by carlescufi