zephyrproject-rtos/zephyr #49942: Support multiple device trees that can be selected at runtime

johnkjellberg, 2022-09-05:
Since it is quite common that components need to be replaced it is must be possible to support different hw revisions. The current Zephyr model handles this by having different BOARD definitions and build separate binaries for each hardware revision. This is not very convenient, since it might just be a small LED driver or other minor difference for example that have changed for example.

I would like to be able to define multiple device trees in one binary that can be selected runtime during boot (by for example checking version pins connected to GPIO).

henrikbrixandersen, 2022-09-05:
You do realise that in Zephyr, the devicetree is not part of the produced binary at all? The devicetree is parsed and a set of C defines are produced, which are used as compile-time constants?

What you are suggesting is a totally different approach more alike what Linux does (where the devicetree is compiled to a devicetree blob, which is parsed at run-time).

johnkjellberg, 2022-09-05:
> You do realise that in Zephyr, the devicetree is not part of the produced binary at all? The devicetree is parsed and a set of C defines are produced, which are used as compile-time constants?

Yes. I'm talking about the result that is generated from the tree. For example the arrays used to initialize the drivers. That part should not be that hard, but there might be other things that is generated that I'm not aware of. I'm open to other suggestions to solve the described problem.

pdgendt, 2022-09-06:
Food for thought; I have a similar use case where a board has GPIO revision indicators.

As already mentioned, it's not an easy task to support multiple devicetrees, but it should be possible to allow some "runtime customization" in a single `dts` file.
My idea was to extend the `hwinfo` API, since it's single instance, with `int hwinfo_get_device_revision(uint32_t *rev)`. Adding a compatible that would implement such a function could look like:
```
gpio_revision {
	compatible = "zephyr,gpio-revision";
	gpios = <&gpio1 0 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>,
		<&gpio1 1 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>,
		<&gpio1 2 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>;
};
```

And create a dependency with other devices, example for `i2c` address change:
```
&i2c0 {
	sensor_old: some_sensor@0 {
		compatible = "vendor,chip";
		reg = <0x00>;
		hwinfo-revision = <0x01>;
	};

	sensor_new: some_sensor@1 {
		compatible = "vendor,chip";
		reg = <0x01>;
		hwinfo-revision = <0x02>, <0x03>, <0x04>;
	};
}
```

Note that this would also require modifications to the device initialization.

pdgendt, 2022-09-07:
I've created an example branch for the `zephyr,gpio-revision` driver [here](https://github.com/pdgendt/zephyr/commits/hwinfo-gpio-revision).