zephyrproject-rtos/zephyr #49953: stm32 gpio_basic_api test fail with CONFIG_ZTEST_NEW_API

FRASTM, 2022-09-06:
**Describe the bug**
when running the  tests/drivers/gpio/gpio_basic_api/drivers.gpio.2pin  on several stm32 target
the testcase fails 

Please also mention any information which could help others to understand
the problem you're facing:
 - target platform:  disco_l475_iot1, nucleo_f103rb, nucleo_f207zg, nucleo_f429zi, nucleo_f746zg, nucleo_g071rb, nucleo_g474re, nucleo_l152re, nucleo_wb55rg, nucleo_wl55jc, stm32f3_disco

 - What have you tried to diagnose or workaround this issue?
 This  faailure comes with the introduction of the  CONFIG_ZTEST_NEW_API    5comit 9422fdff780dba817d9e531de8186af883710ca

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2022-09-06:
@enjiamai could you please give any indication to help fixing this issue ?

enjiamai, 2022-09-06:
Hi @FRASTM , I am looking at it, feedback you soon.

FRASTM, 2022-09-06:
I see a similar pb occurs on `west build -p auto -b disco_l475_iot1 tests/drivers/flash `with commit f2cbd8405b884c7e06dc571a1b070e14cdaeb067  (PR https://github.com/zephyrproject-rtos/zephyr/pull/49502)
```
*** Booting Zephyr OS build zephyr-v3.1.0-4714-ge5b049249eab  ***
Running TESTSUITE flash_driver
===================================================================
E: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x20001678
E: r0/a1:  0x5d305b29  r1/a2:  0x29292929  r2/a3:  0x20646900
E: r3/a4:  0x2828203c r12/ip:  0x657a6973 r14/lr:  0x2029745f
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x08003f90
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x200002c0 (main)
E: Halting system
```

enjiamai, 2022-09-06:
I think I found the problem, my bad, the test suite name is not correct for "mgmt" and "vari". Working on the fix, thanks.

PR #49964 (2022-09-19): tests: driver: gpio: fix the incorrect testsuite names
There are some erros in previous ztest API migration of
the gpio_basic_api tests. Fix the incorrect testsuite
name of the callback mgmt and vari tests.

Fixes #49953
commit 4c9faf95 (2022-09-19): tests: driver: gpio: fix the incorrect testsuite names

There are some erros in previous ztest API migration of
the gpio_basic_api tests. Fix the incorrect testsuite
name of the callback mgmt and vari tests.

Fixes: #49953
closed 2022-09-19 by fabiobaltieri