zephyrproject-rtos/zephyr #49956: `NRF_DRIVE_S0D1` option is not always overriden in the `nordic,nrf-twi` and `nordic,nrf-twim` nodes

IvanVnucec, 2022-09-06:
This statement is not true:
https://github.com/zephyrproject-rtos/zephyr/blob/42858bebb7290e51b93620529bcc1d8b6e5e734b/dts/bindings/pinctrl/nordic%2Cnrf-pinctrl.yaml#L114

Explanation:
I'm using TWI on nrf52832 on custom board.<br>
I have custom board pin control dtsi file `*-pinctrl.dtsi` file which has `i2c` configuration as:
```
[...]

	i2c1_default: i2c1_default {
		group1 {
			psels = <NRF_PSEL(TWIM_SCL, 0, 16)>,
					<NRF_PSEL(TWIM_SDA, 0, 15)>;
		};
	};

	i2c1_sleep: i2c1_sleep {
		group1 {
			psels = <NRF_PSEL(TWIM_SCL, 0, 16)>,
					<NRF_PSEL(TWIM_SDA, 0, 15)>;
			low-power-enable;
		};
	};

[...]
```
And I use this `*-pinctrl.dtsi` file in my custom board dtsi file as:
```
#include "art_nr1amp-pinctrl.dtsi"

[...]

&i2c1 {
	compatible = "nordic,nrf-twi";
	status = "okay";
	pinctrl-0 = <&i2c1_default>;
	pinctrl-1 = <&i2c1_sleep>;
	pinctrl-names = "default", "sleep";
};

[...]
```
On my custom board I have external pull-up resistors on both SCL and SDA lines. 
When I try to interface I2C sensor with this code:
```c
#include <zephyr/zephyr.h>
#include <zephyr/drivers/i2c.h>

void main(void)
{
	const struct device *const i2c_dev = DEVICE_DT_GET(DT_NODELABEL(i2c1));

	if (!device_is_ready(i2c_dev)) {
		printk("ERROR: I2C device is not ready.\n");
		return;
	}

	int retval;
	while (1) {
		uint8_t temp[2];
		retval = i2c_burst_read(i2c_dev, 0x40, 0xE3, temp, sizeof(temp));
		if (retval == 0) {
                        [...]
		} else {
			printk("read register err: %d\n", retval);
		}

		k_msleep(500);
	}
}
```
The code breaks at `i2c_burst_read` function and I get `read register err: -5` error code on the console output.
I've looked at the SDA and SCL lines with the oscilloscope and on the Slave ACK there is Vdd/2 signal on the SDA line (should be Vdd). I've interpreted the error that I have internal pull-up resistor enabled.<br>
To disable the internal pull-up I've added `nordic,drive-mode = < NRF_DRIVE_S0D1  >;` line in my custom board pin control file so it became:
```
[...]

	i2c1_default: i2c1_default {
		group1 {
			psels = <NRF_PSEL(TWIM_SCL, 0, 16)>,
					<NRF_PSEL(TWIM_SDA, 0, 15)>;
                        nordic,drive-mode = < NRF_DRIVE_S0D1  >;
		};
	};

	i2c1_sleep: i2c1_sleep {
		group1 {
			psels = <NRF_PSEL(TWIM_SCL, 0, 16)>,
					<NRF_PSEL(TWIM_SDA, 0, 15)>;
                        nordic,drive-mode = < NRF_DRIVE_S0D1  >;
			low-power-enable;
		};
	};

[...]
```
And the had `i2c_burst_read` function successfully returned `0` and I've could successfully retrieved data from the I2C sensor.

henrikbrixandersen, 2022-09-06:
> And the had `i2c_burst_read` function successfully returned `0` and I've could successfully retrieved data from the I2C sensor.

So what are you asking here?

IvanVnucec, 2022-09-06:
> > And the had `i2c_burst_read` function successfully returned `0` and I've could successfully retrieved data from the I2C sensor.
> 
> So what are you asking here?

I've edited the end of the issue to clarify it.

marc-cpdesign, 2022-09-11:
was this issue fixed in 32a87215d7aa97191097357f602237f7da4a3536 ?

IvanVnucec, 2022-09-11:
I've edited comment with information that I'm developing software with the nrf sdk v2.0.2. Zephyr version on sdk could be found [here](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/releases/release-notes-2.0.2.html#zephyr).

anangl, 2022-09-19:
> So the statement linked at the start of this issue might be false. Can someone verify it?

Yes, this is a bug in NCS v2.0.2. A workaround for it is to set `nordic,drive-mode = <NRF_DRIVE_S0D1>` in the pinctrl configuration node, just as you did (there is no need to set it in the `*_sleep` node, as the drive mode is irrelevant in this case - the pins are not driven anyway).

The fix mentioned above (32a8721) is now also applied in NCS, in both `main` (the recently released v2.1.0 contains this fix) and `v2.0-branch` branches, hence I'm closing this issue.

closed 2022-09-19 by anangl