zephyrproject-rtos/zephyr #49985: STM32:NUCLEO_WL55JC No serial RX in STOP mode

yourapiexpert, 2022-09-06:
**Describe the bug**
I am trying to contribute a new overlay file for samples/boards/stm32/power_mgmt/serial_wakeup to help folks with the Nucleo_WL55JC but I have hit a snag. Following the previous examples I have a overlay as follows:

```
&clk_lse {
	status = "okay";
};

&clk_hsi {
	/* Comment out this line to use HSI as clk source */
	status = "disabled";
};

&clk_lsi {
	/* Comment out this line to use LSI as clk source */
	status = "disabled";
};

&lptim1 {
	clocks = <&rcc STM32_CLOCK_BUS_APB1 0x80000000>,
		 <&rcc STM32_SRC_LSE LPTIM1_SEL(3)>;
	status = "okay";
};

&clk_msi {
	status = "okay";
	msi-range = < 6 >;
	msi-pll-mode;
};

&pll {
	div-m = <1>;
	mul-n = <24>;
	div-p = <4>;
	div-r = <2>;
	div-q = <2>;
	clocks = <&clk_msi>;
	status = "okay";
};

&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(48)>;
	cpu1-prescaler = <1>;
	cpu2-prescaler = <1>;
	ahb3-prescaler = <1>;
	apb1-prescaler = <1>;
	apb2-prescaler = <1>;
};

&lpuart1 {
	pinctrl-0 = <&lpuart1_tx_pa2 &lpuart1_rx_pa3>;
	pinctrl-1 = <&analog_pa2 &analog_pa3>;
	pinctrl-names = "default","sleep";
	clocks = <&rcc STM32_CLOCK_BUS_APB1_2 0x00000001>,
	<&rcc STM32_SRC_LSE LPUART1_SEL(3)>;
	current-speed = <9600>;
	wakeup-source;
	status = "okay";
};
```

When running the example in samples/boards/stm32/power_mgmt/serial_wakeup I successfully get output on the console but the device fails to wake up. If I hold 'tab' whilst it is booting I can get output from the shell until I stop and the device goes to sleep never to wake again.

**To Reproduce**
Steps to reproduce the behavior:
1. Create samples/boards/stm32/power_mgmt/serial_wakeup/boards/nucleo_wl55jc.overlay with the info from above
2. west build --pristine -b nucleo_wl55jc samples/boards/stm32/power_mgmt/serial_wakeup/
4. See error

**Expected behavior**
The device to wake from its slumber and the shell/console to respond to input.

**Impact**
Medium. I can get around it be not enabling CONFIG_PM but it doesn't help my battery life at all.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr 0.15.0
 - Commit: latest, freshest main

FRASTM, 2022-09-08:
@yourapiexpert could you test including PR https://github.com/zephyrproject-rtos/zephyr/pull/50008 (mainly commit 4a163618f30e6bcf4aaccb77eb745f42f31eec31 ) and   PR https://github.com/zephyrproject-rtos/zephyr/pull/50045

yourapiexpert, 2022-09-09:
@FRASTM It works so much better. Thank you and @erwango for all the efforts.

In my case I think there might be some unintended consequences.  In my testing:

1. I have to set CONFIG_SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN=n otherwise it's sluggish and slow to respond.  With interrupt it works, eventually, it's just not responsive.

2. If I set CONFIG_COUNTER=y then everything stops. There is no output or activity.

This is using the stock serial_wakeup sample and the dts you posted in PR #50045.

But, with those two turned off it's going!

erwango, 2022-09-09:
Reopening, as it's not fully fixed.

FRASTM, 2022-09-09:
yes, this seems related to the PM of the stm32wl because  without cpu-power-states defined (but CONFIG_PM), the console reacts as quick as stm32wb (where cpu-power-states are defined and running on the LPUART too)

yourapiexpert, 2022-09-09:
Here's what I have at least for the lpuart:

```
&lpuart1 {
	pinctrl-1 = <&analog_pa2 &analog_pa3>;
	pinctrl-names = "default","sleep";
	/delete-property/ clocks;
	clocks = <&rcc STM32_CLOCK_BUS_APB1_2 0x00000001>,
		 <&rcc STM32_SRC_LSE LPUART1_SEL(3)>;
	wakeup-source;
	current-speed = <9600>;
};
```
The default dts for the wl55jc has the CPU states but it's still sluggish.  Well not to much sluggish as more responds to about every second character.

FRASTM, 2022-09-09:
I found that a config is missing to wake up by the LPUART  before entering to stop mode.
PR to come soon

yourapiexpert, 2022-09-09:
@FRASTM That has completely fixed it! I can confirm that when run with this example and correct configs applied there is no more sluggish behavior.

I can also report that the issue with CONFIG_LPTIM or the low power timer config is no longer present.

There is still a collision with anything CONFIG_COUNTER related even after PR #50104 is applied but I think that is another matter.

Thanks so much for all the help and support.

erwango, 2022-09-12:
Re-opening as #50100 will be required for completing the fix.

> There is still a collision with anything CONFIG_COUNTER related

Ok, please raise another issue.

closed 2022-09-15 by fabiobaltieri