zephyrproject-rtos/zephyr #49987: Unable to compile on windows

vincemulhollon, 2022-09-06:
**Describe the bug**

Unable to compile samples for multiple boards on Windows using a Python venv following the instructions from:

https://docs.zephyrproject.org/latest/develop/getting_started/index.html

I've tried deleting my .cache and build dirs, no change
I've tried cmake versions 3.20.0 and 3.24.0 same error
I've tried compiling for a nucleo_f767zi and an esp32 same error
I've tried compiling samples\basic\blinky and samples\basic\button same error.

**To Reproduce**

Follow the Windows steps using the virtual environment option as seen at:

https://docs.zephyrproject.org/latest/develop/getting_started/index.html

There is one tiny exception to the documented process.  During the "pip install -r %HOMEPATH%\zephyrproject\zephyr\scripts\requirements.txt" step, the installation of Python's psutil will fail with an error message  “Microsoft Visual C++ 14.0 or greater is required. Get it with Microsoft C++ Build Tools”, so I installed VC++ build tools, reran the pip install process, everything installed error free at that point.  This is documented as normal behavior for psutil: https://github.com/giampaolo/psutil/blob/master/INSTALL.rst

**Expected behavior**

samples\basic\blinky should compile on a freshly installed machine without a "FATAL ERROR"

**Impact**

Being unable to compile anything on Windows would kind of be a showstopper, on Windows.  I have successfully used Zephyr on Ubuntu, which is a superior operating  system.

**Logs and console output**

The exit code corruption overwriting the ...tatic library line is as displayed on my screen. 

(.venv) C:\Users\vince\zephyrproject\zephyr>west build -b nucleo_f767zi samples\basic\blinky -p
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/vince/zephyrproject/zephyr/samples/basic/blinky
-- Found Python3: C:/Users/vince/zephyrproject/.venv/Scripts/python.exe (found suitable exact version "3.10.2") found components: Interpreter
-- Cache files will be written to: C:/Users/vince/zephyrproject/zephyr/.cache
-- Zephyr version: 3.1.99 (C:/Users/vince/zephyrproject/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: nucleo_f767zi
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.0 (C:/Users/vince/zephyr-sdk-0.15.0)
-- Found toolchain: zephyr 0.15.0 (C:/Users/vince/zephyr-sdk-0.15.0)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Users/vince/zephyrproject/zephyr/boards/arm/nucleo_f767zi/nucleo_f767zi.dts
-- Generated zephyr.dts: C:/Users/vince/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/vince/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/vince/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing C:/Users/vince/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/vince/zephyrproject/zephyr/boards/arm/nucleo_f767zi/nucleo_f767zi_defconfig'
Merged configuration 'C:/Users/vince/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to 'C:/Users/vince/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Users/vince/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/vince/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/vince/zephyrproject/zephyr/build
-- west build: building application
[3/158] Generating include/generated/version.h
-- Zephyr version: 3.1.99 (C:/Users/vince/zephyrproject/zephyr), build: zephyr-v3.1.0-4786-ge9e7ad388da7
Exit code 0xc0000417tatic library zephyr\arch\arch\arm\core\aarch32\libarch__arm__core__aarch32.a

CMake Error: Generator: execution of make failed. Make command was: C:/ProgramData/chocolatey/bin/ninja.exe &&
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\vince\zephyrproject\zephyr\build'

(.venv) C:\Users\vince\zephyrproject\zephyr\build>C:/ProgramData/chocolatey/bin/ninja.exe
[0/137] Linking C static library zephyr\arch\arch\arm\core\aarch32\libarch__arm__core__aarch32.a
(.venv) C:\Users\vince\zephyrproject\zephyr\build>

**Environment (please complete the following information):**

Windows 11

**Additional context**

I looked around for open issues and did not seem to find anything applicable.

My theory is some requirement is not present.  But I'm not sure where to find in the logs or build process what is missing other than I can see the "Exit code 0xc0000417" being displayed on the screen.  Interesting that ESP32 and a STM32 build fail the same way (while clearing .cache and \build in between of course)

carlescufi, 2022-09-06:
> 'Exit code 0xc0000417'

It looks like a process crashed. Likely `cmake` or `gcc`? @stephanosio FYI

carlescufi, 2022-09-06:
I have not been able to reproduce it:

```
C:\Users\carles\src\zephyr\zephyr>west build -b nucleo_f767zi samples\basic\blinky
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/carles/src/zephyr/zephyr/samples/basic/blinky
-- Found Python3: C:/Python310/python.exe (found suitable exact version "3.10.5") found components: Interpreter
-- Cache files will be written to: C:/Users/carles/src/zephyr/zephyr/.cache
-- Zephyr version: 3.1.99 (C:/Users/carles/src/zephyr/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: nucleo_f767zi
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.0 (C:/Users/carles/bin/zephyr-sdk-0.15.0)
-- Found toolchain: zephyr 0.15.0 (C:/Users/carles/bin/zephyr-sdk-0.15.0)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Users/carles/src/zephyr/zephyr/boards/arm/nucleo_f767zi/nucleo_f767zi.dts
-- Generated zephyr.dts: C:/Users/carles/src/zephyr/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/carles/src/zephyr/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/carles/src/zephyr/zephyr/build/zephyr/dts.cmake
Parsing C:/Users/carles/src/zephyr/zephyr/Kconfig
Loaded configuration 'C:/Users/carles/src/zephyr/zephyr/boards/arm/nucleo_f767zi/nucleo_f767zi_defconfig'
Merged configuration 'C:/Users/carles/src/zephyr/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to 'C:/Users/carles/src/zephyr/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Users/carles/src/zephyr/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/carles/bin/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/carles/src/zephyr/zephyr/build
-- west build: building application
[1/158] Generating include/generated/version.h
-- Zephyr version: 3.1.99 (C:/Users/carles/src/zephyr/zephyr), build: zephyr-v3.1.0-4877-gc75e6cfcb939
[148/158] Linking C executable zephyr\zephyr_pre0.elf

[152/158] Linking C executable zephyr\zephyr_pre1.elf

[158/158] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       15444 B         2 MB      0.74%
            SRAM:        4480 B       384 KB      1.14%
            DTCM:          0 GB       128 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%

C:\Users\carles\src\zephyr\zephyr>
```

stephanosio, 2022-09-07:
Unable to reproduce this either. Likely a local issue on your system (maybe corrupted Ninja executable?)

carlescufi, 2022-09-07:
Closing since it's hard to see any fault attributable to Zephyr's toolchain or code here. @vincemulhollon please reinstall everything and try again.

closed 2022-09-07 by carlescufi

vincemulhollon, 2022-09-09:
The above issue #50017 contains a working fix, its an unusual anti-virus incompatibility.