zephyrproject-rtos/zephyr #49995: Support for other PHYs than KSZ8081 missing in mcux_eth / nxp kinetis ethernet driver

nelarsen, 2022-09-07:
The Zephyr shim driver eth_mcux.c is currently hard-coded with support for only one particular PHY (KSZ8081, found on many NXP evks).

Quote by @danieldegrasse in [discussion about this issue at Discord](https://discord.com/channels/720317445772017664/1012637723347124224/1014198439824851004):
> The PHY management needs to be moved out of the ethernet driver, and into a proper PHY driver. There is a PHY driver interface present in Zephyr, but NXP's driver does not currently use it for the PHY

This issue is a show-stopper for any NXP SoC (K6x, RT10xx, RT11xx) using the Kinetis Ethernet driver (eth_mcux) on a board with a PHY different than KSZ8081.

For example, we use the Texas Instruments TLK111 PHY. Only a few modifications of eth_mcux.c (change hard-coded KSZ8081 registers to TLK111) are necessary to get it to work, but **_there is no obvious and trivial way to generalize this without refactoring the eth_mcux driver_** like @danieldegrasse suggests in the quote.

igalloway, 2023-06-14:
Bump that we are in the same situation in supporting TJA1103 on FMURT6, FMURT7 and MR-CANHUBK344

decsny, 2023-08-02:
> Bump that we are in the same situation in supporting TJA1103 on FMURT6, FMURT7 and MR-CANHUBK344

I am redoing the driver downstream and it is a large effort that is taking a long time, but it will be able to support the Zephyr PHY API once it is working and therefore should be much more flexible to work with any PHY

closed 2023-11-28 by dleach02

decsny, 2023-11-28:
closed with merge of new nxp_enet driver, eth_mcux will not support this