zephyrproject-rtos/zephyr #49996: tests: drivers: clock_control: nrf_lf_clock_start and nrf_onoff_and_bt fails

nordic-piks, 2022-09-07:
**Describe the bug**
The tests/drivers/clock_control/nrf_onoff_and_bt/drivers.clock.nrf_onoff_and_bt,
 tests/drivers/clock_control/nrf_lf_clock_start/drivers.clock.nrf_lf_clock_start_rc_no_wait,
tests/drivers/clock_control/nrf_lf_clock_start/drivers.clock.nrf_lf_clock_start_xtal_no_wait fails

Observed at drivers.clock.nrf_onoff_and_bt for:
- nrf52840dk_nrf52840
- nrf52dk_nrf52832

Observed at drivers.clock.nrf_lf_clock_start_rc_no_wait for:
- nrf5340dk_nrf5340_cpunet
- nrf5340dk_nrf5340_cpuapp
- nrf52840dk_nrf52840
- nrf52dk_nrf52832

Observed at drivers.clock.nrf_lf_clock_start_xtal_no_wait for:
- nrf52840dk_nrf52840
- nrf52dk_nrf52832

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/drivers/clock_control/nrf_onoff_and_bt -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2022-09-06T20:24:56.749Z] ERROR   - *** Booting Zephyr OS build zephyr-v3.1.0-4786-ge9e7ad388da7  ***
[2022-09-06T20:24:56.749Z] Running TESTSUITE nrf_onoff_and_bt
[2022-09-06T20:24:56.749Z] ===================================================================
[2022-09-06T20:24:56.749Z] START - test_bt_interrupted
[2022-09-06T20:24:56.749Z] Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/clock_control/nrf_onoff_and_bt/src/main.c:75: check_hf_status: (type not equal to exp_on ? NRF_CLOCK_HFCLK_HIGH_ACCURACY : NRF_CLOCK_HFCLK_LOW_ACCURACY)
[2022-09-06T20:24:56.749Z] 354: Clock expected to be on
[2022-09-06T20:24:56.749Z] FAIL - test_bt_interrupted in 0.231 seconds
[2022-09-06T20:24:56.749Z] ===================================================================
[2022-09-06T20:24:56.749Z] START - test_onoff_interrupted
[2022-09-06T20:24:56.749Z] Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/clock_control/nrf_onoff_and_bt/src/main.c:75: check_hf_status: (type not equal to exp_on ? NRF_CLOCK_HFCLK_HIGH_ACCURACY : NRF_CLOCK_HFCLK_LOW_ACCURACY)
[2022-09-06T20:24:56.749Z] 5: Clock expected to be on
[2022-09-06T20:24:56.749Z] FAIL - test_onoff_interrupted in 0.027 seconds
[2022-09-06T20:24:56.749Z] ===================================================================
[2022-09-06T20:24:56.749Z] TESTSUITE nrf_onoff_and_bt failed.
[2022-09-06T20:24:56.749Z] ------ TESTSUITE SUMMARY START ------
[2022-09-06T20:24:56.749Z] SUITE FAIL -   0.00% [nrf_onoff_and_bt]: pass = 0, fail = 2, skip = 0, total = 2 duration = 0.258 seconds
[2022-09-06T20:24:56.749Z] - FAIL - [nrf_onoff_and_bt.test_bt_interrupted] duration = 0.231 seconds
[2022-09-06T20:24:56.749Z] - FAIL - [nrf_onoff_and_bt.test_onoff_interrupted] duration = 0.027 seconds
[2022-09-06T20:24:56.749Z] ------ TESTSUITE SUMMARY END ------
[2022-09-06T20:24:56.749Z] ===================================================================
[2022-09-06T20:24:56.749Z] RunID: 94f61b6252ac82c746e72c0a4c4ef876
[2022-09-06T20:24:56.749Z] PROJECT EXECUTION FAILED
```

```
[2022-09-06T20:25:41.640Z] ERROR   - Running TESTSUITE nrf_lf_clock_start
[2022-09-06T20:25:41.640Z] ===================================================================
[2022-09-06T20:25:41.640Z] CLOCK_CONTROL_NRF_K32SRC=RC
[2022-09-06T20:25:41.640Z] SYSTEM_CLOCK_NO_WAIT=y
[2022-09-06T20:25:41.640Z] SYSTEM_CLOCK_WAIT_FOR_AVAILABILITY=n
[2022-09-06T20:25:41.640Z] SYSTEM_CLOCK_WAIT_FOR_STABILITY=n
[2022-09-06T20:25:41.640Z] START - test_clock_check
[2022-09-06T20:25:41.640Z] Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/clock_control/nrf_lf_clock_start/src/main.c:32: rc_check: (on is true)
[2022-09-06T20:25:41.640Z] Clock should be off
[2022-09-06T20:25:41.640Z] FAIL - test_clock_check in 0.016 seconds
[2022-09-06T20:25:41.640Z] ===================================================================
[2022-09-06T20:25:41.640Z] START - test_wait_in_thread
[2022-09-06T20:25:41.640Z] PASS - test_wait_in_thread in 0.001 seconds
[2022-09-06T20:25:41.640Z] ===================================================================
[2022-09-06T20:25:41.640Z] TESTSUITE nrf_lf_clock_start failed.
[2022-09-06T20:25:41.640Z] ------ TESTSUITE SUMMARY START ------
[2022-09-06T20:25:41.640Z] SUITE FAIL -  50.00% [nrf_lf_clock_start]: pass = 1, fail = 1, skip = 0, total = 2 duration = 0.017 seconds
[2022-09-06T20:25:41.640Z] - FAIL - [nrf_lf_clock_start.test_clock_check] duration = 0.016 seconds
[2022-09-06T20:25:41.640Z] - PASS - [nrf_lf_clock_start.test_wait_in_thread] duration = 0.001 seconds
[2022-09-06T20:25:41.640Z] ------ TESTSUITE SUMMARY END ------
[2022-09-06T20:25:41.640Z] ===================================================================
[2022-09-06T20:25:41.640Z] RunID: e848f3e15f070abdbf52382330b80c20
[2022-09-06T20:25:41.640Z] PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.14.1
 - Commit SHA or Version used: v3.1.0-4786-ge9e7ad388da7

PerMac, 2022-10-04:
@nordic-krch Any progress with this one?

anangl, 2022-11-28:
The problem with drivers.clock.nrf_onoff_and_bt test was fixed by #52454. nrf_lf_clock_start* tests will be fixed by #52333.

PR #52333 (2022-11-29): tests: drivers: nrf_lf_clock_control: Fix test initialization 
This test needs an initialization step to be done soon after the system
clock is started. ZTEST setup function is too late for that, so use an
initialization function executed at the POST_KERNEL stage instead.

Fixes #49996.
closed 2022-11-29 by carlescufi