zephyrproject-rtos/zephyr #50028: flash_stm32_ospi Write enable failed when building with TF-M

tobias-aunbol, 2022-09-08:
**Describe the bug**
On b_u585_iot02a:
When enabling Trustzone with CONFIG_BUILD_WITH_TFM=y, the flash_stm32_ospi driver will fail in initializing the MX25LM51245 external OSPI flash giving following errors:

```
[00:00:05.102,000] <err> flash_stm32_ospi: OSPI config auto polling failed
[00:00:05.102,000] <err> flash_stm32_ospi: OSPI write Enable failed
[00:00:05.102,000] <err> flash_stm32_ospi: OSPI mode not config'd (8 rate 2)
```
My first thought was that must be related to something with the configuration in the secure application. I have not made any changes to the secure application provided in the TF-M module. I'm no expert in Trustzone yet, but when I investigate the hal and secure configurations for target provided under stm32u5xx/bl2/boot_hal_bl2.c,  ../low_level_security.c and secure/target_cfg.c i can't find anything that looks like that the OSPI (which is a securable peripheral) should be configured to be only accessible from secure through the GTZC. After the configuration i also have read the status of the GTZC1_TZSC_SECCFGR3 register where it looks like that the only peripheral that should be secure is HASHSEC and RNGSEC, all other is 0's which is corresponding fine with what is initialized in target_cfg.c in the function gtzc_init_cfg. 

I need the OSPI external flash accessible from non-secure, as I'm using littlefs mounted on it, which is running in non-secure. 

So what am i missing? Is it a bug or is there something I misunderstand?
I have raised the question both in st community and on the discord channels both ST and TF-M, also without any luck. I'm bit of doubt where/what this issue is related to.

**To Reproduce**
I also tested the issue when building the tfm_integration/tfm_ipc sample, to do this i added the lfs_cfg node to b_u585i_iot02a_ns.dts:

```

	fstab {
		compatible = "zephyr,fstab";
		lfs_cfg: lfs {
			compatible = "zephyr,fstab,littlefs";
			mount-point = "/lfs";
			partition = <&external_storage_partition>; //This needs to be named in the the mx25lm51245 node in:
                                                                                           //b_u585i_iot02a-common.dtsi                                                                                                                                                            
			read-size = <16>;
			prog-size = <16>;
			cache-size = <64>;
			lookahead-size = <64>;
			block-cycles = <512>;
			automount;
		};
	};
```
In b_u585i_iot02a-common.dtsi in the mxlm25lm51245 node name partition@0 as:

```
partitions {
			   compatible = "fixed-partitions";
			   #address-cells = <1>;
			   #size-cells = <1>;

			   external_storage_partition: partition@0 {
			       label = "nor";
			       reg = <0x00000000 DT_SIZE_M(4)>;
			   };
		};

```

* Add the following to prj.conf for the ipc_sample:

```
# littlefs
CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_LITTLEFS=y

CONFIG_LOG=y #For seeing the error output from ospi driver
```
* Build the sample using: west build -p -b b_u585i_iot02a_ns tfm_ipc
* Run the regression script under ./build/tfm/regression.sh
* flash the sample to target using west flash
* Watch log output 

**Expected behavior**
As OSPI peripheral is set to non-secure it is expected that the OSPI driver is able to initilize and work from non-secure, just as if the application was build without CONFIG_BUILD_WITH_TFM. 

**Impact**
Not able to use external OSPI flash from non-secure, which means that we are not able to build with Trustzone and use the secure features.

**Environment (please complete the following information):**
 - OS: Linux Ubuntu
 - Zephyr SDK 0.15.0
 - SHA: 5c27067dcd9bc03d911aaf5b1628ab56e1c5f4ca

erwango, 2022-09-09:
@tobias-aunbol I'm not able to reproduce the issue you're seeing. Can you check there is no missing configuration bit?
Also can you share the log you're getting ?

tobias-aunbol, 2022-09-12:
@erwango thanks for looking into this. Okay that sounds weird.

Can you verify that the littleFS is actually mounted on yours when building with TF-M?

When i'm building without TF-M and littlefs i get following log output from flash_stm32_ospi driver and the littlefs:

```
[00:00:00.141,000] <inf> flash_stm32_ospi: OSPI flash config is OPI / STR
[00:00:00.141,000] <inf> flash_stm32_ospi: Read SFDP from octoFlash
[00:00:00.141,000] <inf> flash_stm32_ospi: Read SFDP from octoFlash
[00:00:00.145,000] <inf> littlefs: littlefs partition at /lfs
[00:00:00.145,000] <inf> littlefs: LittleFS version 2.4, disk version 2.0
[00:00:00.145,000] <inf> littlefs: FS at ospi-nor-flash@0:0x0 is 1024 0x1000-byte blocks with 512 cycle
[00:00:00.145,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 64
[00:00:00.145,000] <inf> littlefs: /lfs mounted
[00:00:00.145,000] <inf> littlefs: Automount /lfs succeeded
*** Booting Zephyr OS build zephyr-v3.1.0-4747-g938096a6c8c1  **
```
So i would expect that you get the same output if the littlefs is successfully mounted when building with TF-M.

When i'm enabling TF-M (setting the following in prj.conf)
```
CONFIG_BUILD_WITH_TFM=y
CONFIG_TFM_PROFILE_TYPE_NOT_SET=y
CONFIG_TFM_IPC=y
```
And then build with the modified b_u585_iot02_ns.dts (as i mentioned) I get following log output:

```
00:00:05.101,000] <err> flash_stm32_ospi: OSPI config auto polling failed
[00:00:05.101,000] <err> flash_stm32_ospi: OSPI write Enable failed
[00:00:05.101,000] <err> flash_stm32_ospi: OSPI mode not config'd (8 rate 1)
[00:00:05.105,000] <inf> littlefs: littlefs partition at /lfs
[00:00:05.105,000] <inf> littlefs: LittleFS version 2.4, disk version 2.0
[00:00:05.105,000] <err> littlefs: can't open flash area 0
[00:00:05.105,000] <err> fs: fs mount error (-19)
[00:00:05.105,000] <err> littlefs: Automount /lfs failed: -19
*** Booting Zephyr OS build zephyr-v3.1.0-4747-g938096a6c8c1  ***
```

When you talk about missing configuration bit, what register/bit are you talking about here? I'm not making any changes to the driver/application besides enabling TFM.
Here i logged the GTZC registers from secure:
[SEC_GTZC1_TZSC.log](https://github.com/zephyrproject-rtos/zephyr/files/9545728/SEC_GTZC1_TZSC.log)

When i debug the issue, i can see that when building with TF-M the flash_ospi_driver is failing inside HAL_OSPI_AutoPolling where after WRITE_REG(hospi->Instance->IR, ir_reg);
The busy flag never goes down leaving the status register in the following state:

(gdb) p/t *0x420d2420
$8 = 100100

erwango, 2022-09-12:
> When i'm enabling TF-M (setting the following in prj.conf)

Which sample are you actually using? 
You mentioned `tfm_ipc` initially, but on this sample, this is the default configuration and w/o it this can't build, so I guess this is not what you are using.
Also I wonder why you're not puttin `fstab` node in common.dtsi file if you want to use it in both TFM enabled/ns and TFM disabled applications.

tobias-aunbol, 2022-09-13:
@erwango okay i'm sorry for the missing clarity, i'm primarly using my own application with our own custom boards, I have only provided/tested it with the tfm_ipc example and the b_u585i_iot02a/_ns board so i was able to make it reproducible, thats also why it may become a bit unclear, sorry about that. When i'm testing with the tfm_ipc sample without the TF-M enabled i am course outcommenting everything related to tfm and building for the b_u585i_iot02 board. And because the lfs node is set to automount this will just mount the filesystem on boot. So the tfm_ipc sample could in this case in fact just be an empty application with the same configurations as I provided. 

So to sum up you should be able to take whatever application, enabling TF-M,  have the lfs node set to automount, let the partition phandle point to the partition in the mxlm25lm51245 node, and build, and the flash_stm32_ospi driver should fail in HAL_OSPI_AutoPolling. And when you disabling the TF-M in prj.conf and build for the same board without the _ns  prefix you should be to see that the lfs is successfully mounted. Just as my logs shows. 

Hope this is a bit more clear, or else let me now :)

erwango, 2022-09-16:
@tobias-aunbol We managed to get it working. Issue is finally located in TFM, which was not releasing GPIOI correctly.
In https://github.com/zephyrproject-rtos/trusted-firmware-m/blob/4c984817d861512edaeca27b83308e9b9915a98a/platform/ext/target/stm/common/stm32u5xx/secure/target_cfg.c.
In function `pinmux_init_cfg`, need to add :
```
  __HAL_RCC_GPIOI_CLK_ENABLE();
  GPIOI_S->SECCFGR = 0x0;
```
@microbuilder What is the process to fix an issue in Zephyr TF-M fork? Can we push patches in the fork in parallel as in TF-M main repo ?

microbuilder, 2022-09-16:
You file a change request to TF-M, and then we can cherry pick that in the Zephyr fork.

tobias-aunbol, 2022-09-19:
@erwango i just tested, and it looks like it's working properly, filesystem is mounting on external flash and i'm able to read and write! Thanks a lot!

erwango, 2022-09-20:
@microbuilder Here it is: https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/16791

erwango, 2022-09-22:
I propose to keep the ticket open, so we can track it is cherry-picked in Zephyr TFM fork.

tobias-aunbol, 2022-09-22:
yeah good idea, sorry i pulled the trigger to early ;)

erwango, 2023-01-05:
@microbuilder Any idea when upstream fix will be integrated in Zephyr TF-M clone (or rather: Zephyr TF-M clone updated with recent TF-M version)

microbuilder, 2023-01-05:
> @microbuilder Any idea when upstream fix will be integrated in Zephyr TF-M clone (or rather: Zephyr TF-M clone updated with recent TF-M version)

We apply cherry picked patches when needed. Just raise a PR in the TF-M fork with the cherry-pick hash. See the latest merged commit for an example. The plan was to update to 1.7.0 after the 2.3 tag in Zephyr.

erwango, 2023-01-06:
ok, created https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/82

erwango, 2023-01-31:
I'm closing this as it's delivered in TF-M-1.7

closed 2023-01-31 by erwango