zephyrproject-rtos/zephyr #50032: tests: subsys: shell: shell.core and drivers.can.shell fails at shell_setup

nordic-piks, 2022-09-08:
**Describe the bug**
The tests/subsys/shell/shell/shell.core and  tests/drivers/can/shell/drivers.can.shell fails

Observed for:
- nrf52840dk_nrf52840
- nrf52dk_nrf52832
- nrf9160dk_nrf9160
- nrf9160dk_nrf9160_ns
- nrf5340dk_nrf5340_cpuapp
- nrf5340dk_nrf5340_cpuapp_ns
- nrf5340dk_nrf5340_cpunet

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/subsys/shell/shell -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2022-09-07T22:19:07.153Z] ERROR   - *** Booting Zephyr OS build zephyr-v3.1.0-4878-g897ae4a2d5df  ***
[2022-09-07T22:19:07.153Z] Running TESTSUITE shell
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/shell/shell/src/main.c:498: shell_setup: (shell_ready(sh) is false)
[2022-09-07T22:19:07.153Z] timed out waiting for dummy shell backend
[2022-09-07T22:19:07.153Z] START - test_cmd_clear
[2022-09-07T22:19:07.153Z] FAIL - test_cmd_clear in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_cmd_dict
[2022-09-07T22:19:07.153Z] FAIL - test_cmd_dict in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_cmd_history
[2022-09-07T22:19:07.153Z] FAIL - test_cmd_history in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_cmd_mand_1_opt_raw
[2022-09-07T22:19:07.153Z] FAIL - test_cmd_mand_1_opt_raw in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_cmd_resize
[2022-09-07T22:19:07.153Z] FAIL - test_cmd_resize in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_cmd_select
[2022-09-07T22:19:07.153Z] FAIL - test_cmd_select in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_cmd_shell
[2022-09-07T22:19:07.153Z] FAIL - test_cmd_shell in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_mand_2_opt_raw
[2022-09-07T22:19:07.153Z] FAIL - test_mand_2_opt_raw in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_max_argc
[2022-09-07T22:19:07.153Z] FAIL - test_max_argc in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_section_cmd
[2022-09-07T22:19:07.153Z] FAIL - test_section_cmd in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_set_root_cmd
[2022-09-07T22:19:07.153Z] FAIL - test_set_root_cmd in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_shell_fprintf
[2022-09-07T22:19:07.153Z] FAIL - test_shell_fprintf in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_shell_module
[2022-09-07T22:19:07.153Z] FAIL - test_shell_module in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_shell_wildcards_dynamic
[2022-09-07T22:19:07.153Z] FAIL - test_shell_wildcards_dynamic in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] START - test_shell_wildcards_static
[2022-09-07T22:19:07.153Z] FAIL - test_shell_wildcards_static in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] TESTSUITE shell failed.
[2022-09-07T22:19:07.153Z] Running TESTSUITE shell_1cpu
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/shell/shell/src/main.c:498: shell_setup: (shell_ready(sh) is false)
[2022-09-07T22:19:07.153Z] timed out waiting for dummy shell backend
[2022-09-07T22:19:07.153Z] START - test_cmd_help
[2022-09-07T22:19:07.153Z] FAIL - test_cmd_help in 0.001 seconds
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] TESTSUITE shell_1cpu failed.
[2022-09-07T22:19:07.153Z] ------ TESTSUITE SUMMARY START ------
[2022-09-07T22:19:07.153Z] SUITE FAIL -   0.00% [shell]: pass = 0, fail = 15, skip = 0, total = 15 duration = 0.015 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_cmd_clear] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_cmd_dict] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_cmd_history] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_cmd_mand_1_opt_raw] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_cmd_resize] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_cmd_select] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_cmd_shell] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_mand_2_opt_raw] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_max_argc] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_section_cmd] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_set_root_cmd] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_shell_fprintf] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_shell_module] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_shell_wildcards_dynamic] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell.test_shell_wildcards_static] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] SUITE FAIL -   0.00% [shell_1cpu]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] - FAIL - [shell_1cpu.test_cmd_help] duration = 0.001 seconds
[2022-09-07T22:19:07.153Z] ------ TESTSUITE SUMMARY END ------
[2022-09-07T22:19:07.153Z] ===================================================================
[2022-09-07T22:19:07.153Z] RunID: 2bb8e2712b4acb012d716e5505d6f355
[2022-09-07T22:19:07.153Z] PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.14.1
 - Commit SHA or Version used: v3.1.0-4878-g897ae4a2d5d

henrikbrixandersen, 2022-09-08:
I think this is correlated with `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=32768`. `WAIT_FOR()` divides this by `USEC_PER_SEC`, which is `1000000` resulting in a `cycle_count` of `0`:

```
#define WAIT_FOR(expr, timeout, delay_stmt)                                                        \
	({                                                                                         \
		uint32_t cycle_count = (sys_clock_hw_cycles_per_sec() / USEC_PER_SEC) * (timeout); \
		uint32_t start = k_cycle_get_32();                                                 \
		while (!(expr) && (cycle_count > (k_cycle_get_32() - start))) {                    \
			delay_stmt;                                                                \
		}                                                                                  \
		(expr);                                                                            \
	})

```

henrikbrixandersen, 2022-09-08:
@nordic-piks Please see #50058

henrikbrixandersen, 2022-09-08:
Setting priority to medium since this will cause a lot of hardware-in-the-loop tests to fail during our v3.2 stabilisation phase.

PR #50058 (2022-09-09): sys: util: fix WAIT_FOR() timeout calculation
Use k_us_to_cyc_ceil32() for calculating the number of hardware cycles for the given timeout value instead of using a custom calculation.
    
Fixes: #50032
commit 86de6b71 (2022-09-09): sys: util: fix WAIT_FOR() timeout calculation

Use k_us_to_cyc_ceil32() for calculating the number of hardware cycles for
the given timeout value instead of using a custom calculation.

Fixes: #50032
closed 2022-09-09 by fabiobaltieri