zephyrproject-rtos/zephyr #50040: shields: Settle on nodelabels naming scheme

erwango, 2022-09-08:
When using shields with boards, it may happen that shields and board both describe use the same nodelabel (to describe the same device: sensor, display, ..), leading to a conflict at device tree compilation.
For instance, same node label 'lsm6dsl' is used for both board and shield description:
https://github.com/zephyrproject-rtos/zephyr/blob/eafc4f875b7443bceffe7f8bee14aa97d7ca4bd6/boards/shields/x_nucleo_iks01a2/x_nucleo_iks01a2.overlay#L28-L30
https://github.com/zephyrproject-rtos/zephyr/blob/eafc4f875b7443bceffe7f8bee14aa97d7ca4bd6/boards/arm/b_l4s5i_iot01a/b_l4s5i_iot01a.dts#L113-L115
This was discussed initially in https://github.com/zephyrproject-rtos/zephyr/issues/48780.

Conclusion of the discussion was to use, in shields device tree description, the following device nodelabel naming scheme:
`<device>_<address>`

Then https://github.com/zephyrproject-rtos/zephyr/pull/50034 was merged using a different approach:
`<device>_<shield_name>`
Reason is that `<device>_<address>` will give conflicts, when using multiple shields using devices at same address but on different busses.
Note that this proposal was initially rejected in #48780 because not concise enough.

Aim of this issue is to settle on agreement that can be applied in all cases, document it and update in-tree shields accordingly

gmarull, 2022-09-08:
I think it makes sense

erwango, 2022-09-09:
@MaureenHelm  ?

MaureenHelm, 2022-09-09:
Agree this makes sense

erwango, 2022-09-09:
> Agree this makes sense

ok, thanks. I'll update

erwango, 2022-10-18:
Closing, as #50526 was merged

closed 2022-10-18 by erwango