zephyrproject-rtos/zephyr #50044: reel_board: pyocd.yaml causes flashing error on reel board

smrtos, 2022-09-08:
**Describe the bug**
Build the hello world and flash it onto reel board.
Saw below error:

> 0001896 W Board ID OSHW is not recognized [mbed_board]
> 0002065 I Loading /home/ming/sources/zephyrproject/zephyr/bld_ctf_reel/zephyr/zephyr.hex [load_cmd]
> 0002741 C cannot write registers pc, r0, r1, r2, r9, sp, lr because core #0 is not halted [__main__]
> Traceback (most recent call last):
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/__main__.py", line 161, in run
>     status = cmd.invoke()
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/subcommands/load_cmd.py", line 125, in invoke
>     programmer.program(filename,
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/flash/file_programmer.py", line 170, in program
>     self._loader.commit()
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/flash/loader.py", line 289, in commit
>     perf = builder.program(chip_erase=chipErase,
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/flash/builder.py", line 490, in program
>     sector_erase_count, page_program_time = self._compute_sector_erase_pages_and_weight(fast_verify)
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/flash/builder.py", line 666, in _compute_sector_erase_pages_and_weight
>     self._analyze_pages_with_crc32(fast_verify)
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/flash/builder.py", line 645, in _analyze_pages_with_crc32
>     self._enable_read_access()
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/flash/builder.py", line 259, in _enable_read_access
>     self.flash.init(self.flash.Operation.VERIFY)
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/flash/flash.py", line 252, in init
>     result = self._call_function_and_wait(self.flash_algo['pc_init'],
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/flash/flash.py", line 653, in _call_function_and_wait
>     self._call_function(pc, r0, r1, r2, r3, init)
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/flash/flash.py", line 555, in _call_function
>     self.target.write_core_registers_raw(reg_list, data_list)
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/core/soc_target.py", line 235, in write_core_registers_raw
>     self.selected_core_or_raise.write_core_registers_raw(reg_list, data_list)
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/coresight/cortex_m.py", line 1175, in write_core_registers_raw
>     self._base_write_core_registers_raw(reg_list, data_list)
>   File "/home/ming/.local/lib/python3.10/site-packages/pyocd/coresight/cortex_m.py", line 1189, in _base_write_core_registers_raw
>     raise exceptions.CoreRegisterAccessError(
> pyocd.core.exceptions.CoreRegisterAccessError: cannot write registers pc, r0, r1, r2, r9, sp, lr because core #0 is not halted
> FATAL ERROR: command exited with status 1: pyocd flash --config /home/ming/sources/zephyrproject/zephyr/boards/arm/reel_board/support/pyocd.yaml -e sector -t nrf52840 -f 4000000 /home/ming/sources/zephyrproject/zephyr/bld_ctf_reel/zephyr/zephyr.hex
> FAILED: zephyr/cmake/flash/CMakeFiles/flash /home/ming/sources/zephyrproject/zephyr/bld_ctf_reel/zephyr/cmake/flash/CMakeFiles/flash
> cd /home/ming/sources/zephyrproject/zephyr/bld_ctf_reel && /usr/bin/cmake -DTARGET=flash -DDEPENDENCIES="" -P /home/ming/sources/zephyrproject/zephyr/cmake/flash/check_runner_dependencies.cmake && /usr/bin/cmake -E env /usr/bin/python3.10 -m west flash
> ninja: build stopped: subcommand failed.
> 

If commented out below line, the error is gone.

https://github.com/zephyrproject-rtos/zephyr/blob/8191423950ce0e1a920bec727f08707ad2b34558/boards/arm/reel_board/support/pyocd.yaml#L4

jfischer-no, 2022-09-21:
Only one board is affected and it depends on pyOCD version 0.34.x, set priority to low.

fabiobaltieri, 2022-09-22:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/50128

closed 2022-09-22 by fabiobaltieri