zephyrproject-rtos/zephyr #50061: Bluetooth: Samples: bluetooth_audio_source does not send multiple streams

carlst, 2022-09-08:
**Describe the bug**
`samples/bluetooth/broadcast_audio_source` with `CONFIG_BT_AUDIO_BROADCAST_SRC_STREAM_COUNT=2` fails to send data after sending initial buffers.

Please also mention any information which could help others to understand
the problem you're facing:
 - platform is nRF52840DK
 - commit 1dfc469bd5543ab1a3761aa8cfd8019e33ca6b8f
 - tried previous commits but other issues with multiple ISO channels prevented testing
 - first 2 buffers appear to be sent for both streams. After this, `stream_sent_cb` is only invoked for 1st stream

**To Reproduce**
Steps to reproduce the behavior:
1. Edit `samples/bluetooth/broadcast_audio_source/prj.conf` to contain `CONFIG_BT_AUDIO_BROADCAST_SRC_STREAM_COUNT=2`
2. Add `printk` to `src/main stream_started_cb` and `stream_sent_cb` to display `stream` value each time a data buffer is sent
3. Build for nRF52840DK
4. Run (e.g. using Segger Embedded Studio) with RTT log

**Expected behavior**
Expect `stream_sent_cb` to show data sent for each stream in log.

**Logs and console output**
See attached console log file `broadcast_audio_source_20220908.txt`

[broadcast_audio_source_20220908.txt](https://github.com/zephyrproject-rtos/zephyr/files/9531004/broadcast_audio_source_20220908.txt)

Thalley, 2022-09-09:
Likely a controller issue

cvinayak, 2022-09-10:
@carlst Please try this #50126 and let me know it resolves the issue you face.

carlst, 2022-09-13:
@cvinayak The issue does not occur with the code branch that fixes https://github.com/zephyrproject-rtos/zephyr/pull/50126. Initial testing indicates `stream_sent_cb` is now being called in an alternating fashion for both streams. Will continue to test both broadcaster and receiver with various samples, but I consider this issue resolved. Please close as you see fit.

cvinayak, 2022-09-13:
@carlst Thank you for testing. The implementation on the synchronised receiver is not optimized and also I noticed that after certain number of subevents in the ISO event the PDUs are not received. I.e. on a nRF52833 DK (that I used today), second stream PDUs where not received (controller generates invalid status for the ISO data). This is to do with the window calculations that need some tuning. I will send separate PRs when I have satisfactory fix.

carlst, 2022-09-13:
@cvinayak Ah, that may explain the variations in `bt_iso_recv_info` `flags`  and `len`  0 values (see attached log from `samples/broadcast_audio_sink` on nRF52840DK with printk in `stream_recv_cb`). I was concerned about stability of multi-BIS (audio). I can attest 1 BIS has been rock-solid stable over extended periods of time (at least an hour on nRF52840DK with a PDM mic as source, and routing sink data through USB to a PC audio/speaker).

[broadcast_audio_source_20220913.txt](https://github.com/zephyrproject-rtos/zephyr/files/9559935/broadcast_audio_source_20220913.txt)

cvinayak, 2022-09-14:
@carlst It is surprising that you have the second stream data as much as the first stream on your board.

The occasional invalid data which appear on both streams is unrelated to what I am observing on my board where first stream get valid data and the second gets invalid data (this is apparently due to the listen time tuning required).

Your invalid data could be due to missing PDUs on-air from the broadcaster. It could be that the Extended Advertising and Periodic Advertising PDUs are taking more time causing that to overlap with BIG events that is being transmitted.

carlst, 2022-09-15:
@cvinayak I think we have moved well beyond the issue initially reported. While awaiting any window or other calculation changes, I will continue testing and attempt to gather more specific information on:
- data fidelity (within each stream)
- data loss (e.g. flags: 0x10 and/or len=0)
- Advertising PDU effects

If you have any further advice, please let me know. Thank you for the help.

PR #50126 (2022-09-19): Bluetooth: Controller: Fix leak in Num Completed Packets and ISO Broadcast of multiple streams
Fix a leak in generation of Number of Completed Packets when
more than one handle has acknowledgements to be processed.

The loop exited for the current handle by dequeueing the
node tx element of the next handle leading to lost Number of
Completed Packets.

Restrict the calculated possible Pre-Transmission subevents
count to a maximum of 1 so that the full ISO interval is
not fully occupied by the BIG event.

Fix ISO broadcast implementation to flush stale ISO data for
all BIS streams and like wise  [... cut]
closed 2022-09-19 by carlescufi