zephyrproject-rtos/zephyr #50065: tests: tests/subsys/shell/shell test case fail in daily test on many platforms

hunterhu3000, 2022-09-09:
**Describe the bug**
tests/subsys/shell/shell test case fail in daily test on it8xxx2_evb, mec15xxevb_assy6853, mec172xevb_assy6906, intel_adsp_cavs* and etc.
including the following tests:
```
shell.core.shell.core.cmd_select
shell.core.shell.core.cmd_clear
shell.core.shell.core.shell_wildcards_dynamic
shell.core.shell.core.cmd_mand_1_opt_raw
shell.core.shell.core.mand_2_opt_raw
shell.core.shell.core.set_root_cmd
shell.core.shell.core.cmd_history
shell.core.shell.core.shell_module
shell.core.shell.core.cmd_shell
shell.core.shell.core.max_argc
shell.core.shell.core.cmd_dict
shell.core.shell.core.section_cmd
shell.core.shell.core.shell_fprintf
shell.core.shell.core.shell_wildcards_static
shell.core.shell.core.cmd_resize
shell.core.shell.core.cmd_help
```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? it8xxx2_evb, mec15xxevb_assy6853, mec172xevb_assy6906, intel_adsp_cavs* and etc.
 - Is this a regression? Yes
   specific commit? 491f4dd7

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p it8xxx2_evb --device-testing --west-flash="../../itetool/loader.sh" --device-serial=/dev/ttyUSB0 -T tests/subsys/shell/shell -vv
2. west build -p always -b it8xxx2_evb tests/subsys/shell/shell 

**Expected behavior**
The tests should pass

**Impact**
16 test cases fail in daily test under tests/subsys/shell/shell

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.1.0-4834-g491f4dd701b0  (delayed boot 1ms) ***
DEBUG   - DEVICE: Running TESTSUITE shell
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_cmd_clear
DEBUG   - DEVICE: shell_execute_cmd(clear): 0
DEBUG   - DEVICE: shell_execute_cmd(clear -h): 1
DEBUG   - DEVICE: shell_execute_cmd(clear --help): 1
DEBUG   - DEVICE: shell_execute_cmd(clear dummy): -22
DEBUG   - DEVICE: shell_execute_cmd(clear dummy dummy): -22
DEBUG   - DEVICE: PASS - test_cmd_clear in 0.024 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_cmd_dict
DEBUG   - DEVICE: shell_execute_cmd(dict1 one): 1
DEBUG   - DEVICE: shell_execute_cmd(dict1 two): 2
DEBUG   - DEVICE: shell_execute_cmd(dict2 one): 2
DEBUG   - DEVICE: shell_execute_cmd(dict2 two): 4
DEBUG   - DEVICE: PASS - test_cmd_dict in 0.015 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_cmd_history
DEBUG   - DEVICE: shell_execute_cmd(history): 0
DEBUG   - DEVICE: shell_execute_cmd(history -h): 1
DEBUG   - DEVICE: shell_execute_cmd(history --help): 1
DEBUG   - DEVICE: shell_execute_cmd(history dummy): -22
DEBUG   - DEVICE: shell_execute_cmd(history dummy dummy): -22
DEBUG   - DEVICE: PASS - test_cmd_history in 0.026 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_cmd_mand_1_opt_raw
DEBUG   - DEVICE: shell_execute_cmd(cmd_mand_1_opt_raw aaa "" bbb): 0
DEBUG   - DEVICE: shell_execute_cmd(cmd_mand_1_opt_raw): 0
[... 185 lines cut ...]
DEBUG   - DEVICE: - PASS - [shell.test_max_argc] duration = 0.018 seconds
DEBUG   - DEVICE: - PASS - [shell.test_section_cmd] duration = 0.031 seconds
DEBUG   - DEVICE: - PASS - [shell.test_set_root_cmd] duration = 0.031 seconds
DEBUG   - DEVICE: - FAIL - [shell.test_shell_fprintf] duration = 0.016 seconds
DEBUG   - DEVICE: - PASS - [shell.test_shell_module] duration = 0.037 seconds
DEBUG   - DEVICE: - PASS - [shell.test_shell_wildcards_dynamic] duration = 0.018 seconds
DEBUG   - DEVICE: - PASS - [shell.test_shell_wildcards_static] duration = 0.038 seconds
DEBUG   - DEVICE:
DEBUG   - DEVICE: SUITE PASS - 100.00% [shell_1cpu]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.051 seconds
DEBUG   - DEVICE: - PASS - [shell_1cpu.test_cmd_help] duration = 0.051 seconds
DEBUG   - DEVICE:
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY END ------
DEBUG   - DEVICE:
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: RunID: 0ce9bdcbd2453c3b0b166b957bb213e8
DEBUG   - DEVICE: PROJECT EXECUTION FAILED
DEBUG   - run status: it8xxx2_evb/tests/subsys/shell/shell/shell.core failed
INFO    - 15/15 it8xxx2_evb               tests/subsys/shell/shell/shell.core                FAILED Failed (device 10.585s)
ERROR   - see: /home/ztest/zephyrproject/zephyr/twister-out/it8xxx2_evb/tests/subsys/shell/shell/shell.core/handler.log
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) Ubuntu20.04
 - Toolchain (e.g Zephyr SDK, ...) 0.15.0
 - Commit SHA or Version used 491f4dd7

hunterhu3000, 2022-09-09:
@jakub-uC Could you please help to check this problem? Probably caused by the following commit. Thanks

commit 491f4dd701b043639b2ab680353249c8e151d2ca
Author: Jakub Rzeszutko <jakub.rzeszutko@nordicsemi.no>
Date:   Tue Sep 6 09:50:52 2022 +0200

    shell: update shell tests
    
    Replace hard fixed delay with function shell_ready to wait
    for shell readiness.
    
    Signed-off-by: Jakub Rzeszutko <jakub.rzeszutko@nordicsemi.no>

jakub-uC, 2022-09-09:
@hunterhu3000 
This issue is already investigated. It seems that the problem is with the `WAIT_FOR` macro that needs to be fixed:

https://github.com/zephyrproject-rtos/zephyr/pull/50058#pullrequestreview-1101349518

henrikbrixandersen, 2022-09-09:
Duplicate of #50066 (which is a duplicate of #50032).

closed 2022-09-09 by henrikbrixandersen