zephyrproject-rtos/zephyr #50066: tests: tests/drivers/can/shell failed in daily test on many platforms

hunterhu3000, 2022-09-09:
**Describe the bug**
tests/drivers/can/shell failed in daily test on it8xxx2_evb, mec15xxevb_assy6853, mec172xevb_assy6906

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? it8xxx2_evb, mec15xxevb_assy6853, mec172xevb_assy6906
 - Is this a regression? yes
   specific commit? 2f925b47a 

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p it8xxx2_evb --device-testing --west-flash="../../itetool/loader.sh" --device-serial=/dev/ttyUSB0 -T tests/drivers/can/shell -vv
2. west build -p always -b it8xxx2_evb tests/drivers/can/shell

**Expected behavior**
The test can pass

**Impact**
37 test cases fail in daily test

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.1.0-4843-g2f925b47a7d1  (delayed boot 1ms) ***
DEBUG   - DEVICE: Running TESTSUITE can_shell
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/shell/src/main.c:587: can_shell_setup: (shell_ready(sh) is false)
DEBUG   - DEVICE: timed out waiting for dummy shell backend
DEBUG   - DEVICE: START - test_can_bitrate
DEBUG   - DEVICE: FAIL - test_can_bitrate in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_bitrate_missing_value
DEBUG   - DEVICE: FAIL - test_can_bitrate_missing_value in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_bitrate_sample_point
DEBUG   - DEVICE: FAIL - test_can_bitrate_sample_point in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_dbitrate
DEBUG   - DEVICE: FAIL - test_can_dbitrate in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_dbitrate_missing_value
DEBUG   - DEVICE: FAIL - test_can_dbitrate_missing_value in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_dbitrate_sample_point
DEBUG   - DEVICE: FAIL - test_can_dbitrate_sample_point in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_filter_add_all_options
DEBUG   - DEVICE: FAIL - test_can_filter_add_all_options in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_filter_add_ext_id
DEBUG   - DEVICE: FAIL - test_can_filter_add_ext_id in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_filter_add_ext_id_mask
DEBUG   - DEVICE: FAIL - test_can_filter_add_ext_id_mask in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_filter_add_missing_id
DEBUG   - DEVICE: FAIL - test_can_filter_add_missing_id in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_filter_add_rtr
DEBUG   - DEVICE: FAIL - test_can_filter_add_rtr in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_filter_add_rtr_mask
DEBUG   - DEVICE: FAIL - test_can_filter_add_rtr_mask in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_filter_add_std_id
DEBUG   - DEVICE: FAIL - test_can_filter_add_std_id in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_can_filter_add_std_id_mask
DEBUG   - DEVICE: FAIL - test_can_filter_add_std_id_mask in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
[... 105 lines cut ...]
DEBUG   - DEVICE: - FAIL - [can_shell.test_can_send_fd_brs] duration = 0.001 seconds
DEBUG   - DEVICE: - FAIL - [can_shell.test_can_send_missing_id] duration = 0.001 seconds
DEBUG   - DEVICE: - FAIL - [can_shell.test_can_send_no_data] duration = 0.001 seconds
DEBUG   - DEVICE: - FAIL - [can_shell.test_can_send_rtr] duration = 0.001 seconds
DEBUG   - DEVICE: - FAIL - [can_shell.test_can_send_std_id] duration = 0.001 seconds
DEBUG   - DEVICE: - FAIL - [can_shell.test_can_show] duration = 0.001 seconds
DEBUG   - DEVICE:
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY END ------
DEBUG   - DEVICE:
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: RunID: 9e70f4d407e5db93cee82138ae2791b9
DEBUG   - DEVICE: PROJECT EXECUTION FAILED
DEBUG   - -- west flash: using runner misc-flasher
/home/ztest/zephyrproject/zephyr/twister-out/it8xxx2_evb/tests/drivers/can/shell/drivers.can.shell
/home/ztest/itetool/ite /home/ztest/zephyrproject/zephyr/twister-out/it8xxx2_evb/tests/drivers/can/shell/drivers.can.shell/zephyr/zephyr.bin

DEBUG   - run status: it8xxx2_evb/tests/drivers/can/shell/drivers.can.shell failed
INFO    - 1/1 it8xxx2_evb               tests/drivers/can/shell/drivers.can.shell          FAILED Failed (device 10.348s)
ERROR   - see: /home/ztest/zephyrproject/zephyr/twister-out/it8xxx2_evb/tests/drivers/can/shell/drivers.can.shell/handler.log
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) Ubuntu20.04
 - Toolchain (e.g Zephyr SDK, ...) 0.15.0
 - Commit SHA or Version used 2f925b47a

hunterhu3000, 2022-09-09:
@henrikbrixandersen Could you please help to check this problem first? It looks like caused by the following commit. Thanks.

commit 2f925b47a7d141d9ff4edf1350a76f9f53f018d0 (HEAD)
Author: Henrik Brix Andersen <henrik@brixandersen.dk>
Date:   Tue Sep 6 13:25:54 2022 +0200

    tests: drivers: can: shell: poll for the dummy shell to be ready

    Poll for the dummy shell to become ready and fail the tests if not.

    Signed-off-by: Henrik Brix Andersen <henrik@brixandersen.dk>

henrikbrixandersen, 2022-09-09:
@hunterhu3000 Thank you for reporting this. I was already made aware of this in #50058 and I'm working on a fix.

henrikbrixandersen, 2022-09-09:
Closed as duplicate.

closed 2022-09-09 by henrikbrixandersen