zephyrproject-rtos/zephyr #50070: LoRa: Support on RFM95 LoRa module combined with a nRF52 board

mayank0206, 2022-09-09:
Hi,
Recently I do something about LoRa in nRF52, I want to interface the RFM95 LoRa module with zephyr.
Does zephyr have a plan to support RFM95 LoRa? 

Thank You!

martinjaeger, 2022-09-15:
The RFM95 modules are based on Semtech SX1276 IP and already supported by Zephyr. See below for an example devicetree overlay which I used some time ago. This was for an STM32 based board, but it will also work in a similar way for nRF52 boards.

```
&spi1 {
	lora: sx1276@0 {
		compatible = "semtech,sx1276";
		reg = <0>;
		label = "sx1276";
		reset-gpios = <&gpiob 7 GPIO_ACTIVE_LOW>;
		dio-gpios = <&gpioa 2 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			    <&gpioa 3 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
		power-amplifier-output = "pa-boost";
		spi-max-frequency = <1000000>;
	};
};
```

mayank0206, 2022-11-04:
Hello Martinjaeger,

Thank you for sharing the example devicetree overlay. I'm getting the following error when I use this:

sx12xx_common: GPIO device not ready ��
os: ***** BUS FAULT *****
os:   Instruction bus error
os: r0/a1:  0x00000000  r1/a2:  0x00000001  r2/a3:  0x00000105
os: r3/a4:  0x132300b5 r12/ip:  0x00000000 r14/lr:  0x00012155
os:  xpsr:  0x01000000
os: Faulting instruction address (r15/pc): 0x132300b4
os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
os: Current thread: 0x20000870 (main)
fatal_error: Resetting system

I would appreciate your support in resolving this error.

martinjaeger, 2022-11-04:
Just some guesses:

GPIO device: Make sure that `CONFIG_LORA_INIT_PRIORITY` is larger than `CONFIG_GPIO_INIT_PRIORITY`, which should normally be the case.

Bus fault: Try to increase the system work queue size via Kconfig, if not already done.
https://github.com/zephyrproject-rtos/zephyr/blob/278120acb600aa03a93dbfb11125882bb0765d40/samples/subsys/lorawan/class_a/prj.conf#L9

mayank0206, 2022-11-04:
Now that I've made the suggested changes, I'm not getting a restart issue, but Lora is not working:

Getting below error:

  <err> sx12xx_common: GPIO device not ready )�
  <err> lora_send: SX1276 Device not ready