zephyrproject-rtos/zephyr #50095: ARC revision Kconfigs wrongly mixed with board name

nordicjm, 2022-09-09:
**Describe the bug**
Using the boards in https://github.com/zephyrproject-rtos/zephyr/tree/main/boards/arc/em_starterkit if we compare the `em_starterkit_em7d_defconfig` and `em_starterkit_em7d_v22_defconfig` files we have:
```
CONFIG_SOC_EMSK=y
CONFIG_SOC_EMSK_EM7D=y
CONFIG_BOARD_EM_STARTERKIT=y
CONFIG_BOARD_EM_STARTERKIT_R23=y
```
and
```
CONFIG_SOC_EMSK=y
CONFIG_SOC_EMSK_EM7D=y
CONFIG_BOARD_EM_STARTERKIT=y
CONFIG_BOARD_EM_STARTERKIT_R22=y
```
This suggests that both boards are using an EM7D ARC SOC and there are 2 revisions of the board, however when looking at the files in the soc folder, such as https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arc/snps_emsk/Kconfig.defconfig.em7d we see configuration that apparently is tied to a board version:
```
config NUM_IRQS
	# must be > the highest interrupt number used
	default 38 if BOARD_EM_STARTERKIT_R23
	default 36 if BOARD_EM_STARTERKIT_R22

config ARC_MPU_VER
	default 4 if BOARD_EM_STARTERKIT_R23
	default 2 if BOARD_EM_STARTERKIT_R22

config SYS_CLOCK_HW_CYCLES_PER_SEC
	default 25000000 if BOARD_EM_STARTERKIT_R23
	default 30000000 if BOARD_EM_STARTERKIT_R22
```
This seems more like SoC configuration differences, not configuration for a board itself, and is making it very hard to try and update the soc/arch system for use with sysbuild. Therefore, please fix this so that it correctly uses SOC_* Kconfigs to determine this configuration rather than board names, the board name should have no impact on which Kconfigs are selected from the `soc` directory perspective

**Impact**
Showstopper

**Environment (please complete the following information):**
 - Commit SHA or Version used: 390464ce3bfd4d5053ff0ee58cf2d9b7c1fe5d5e

ruuddw, 2022-09-09:
There's a 1 to 1 relation between the board and the SoC revisions. It can be argued if the options are more board than SoC level features.
It's not really clear to me what the problem is you run into or trying to solve. What do you need to be changed?

nordicjm, 2022-09-12:
Boards are being changed to use SOC_* defines, the files in soc/ will need to set those up (i.e. the boards will select which soc defines to use). So for example, at present, there are 3 different boards all filtering on `BOARD_EM_STARTERKIT_R22` and `BOARD_EM_STARTERKIT_R23`:
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arc/snps_emsk/Kconfig.defconfig.em11d
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arc/snps_emsk/Kconfig.defconfig.em7d
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arc/snps_emsk/Kconfig.defconfig.em9d

This needs to change so that it uses SOC_* defines

closed 2023-02-18 by github-actions[bot]