zephyrproject-rtos/zephyr #50096: tests: drivers:  the gpio_basic_api test cannot be build successfully on bl5340_dvk_cpuapp board

enjiamai, 2022-09-09:
**Describe the bug**
The tests/drivers/gpio/gpio_basic_api cannot be built successfully. 
After bisecting, the bl5340_dvk_cpuapp failed in CI after this commit: https://github.com/zephyrproject-rtos/zephyr/commit/975787fb5e5070add816eee7e7d46e986061ac29.

It seems like the dts related configuration need to adapt to this change. Need board owner help with addressing it. Thanks!

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always tests/drivers/gpio/gpio_basic_api/ -b bl5340_dvk_cpuapp
2. See build error

**Expected behavior**
The test should be built successfully.

**Impact**
The gpio test cannot be built successfully in CI.

**Logs and console output**
```
-- west build: building application
[1/183] Preparing syscall dependency handling

[2/183] Generating include/generated/version.h
-- Zephyr version: 3.1.99 (/home/mm/zephyrproject/zephyr), build: zephyr-v3.1.0-5045-g4625654da38f
[173/183] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/mm/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map
: && ccache /home/mm/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/mm/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/cmse/libarch__arm__core__aarch32__cortex_m__cmse.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/subsys/random/libsubsys__random.a  zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/mm/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/thumb/v8-m.main/nofp"  -L/home/mm/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m33  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/mm/zephyrproject/zephyr/build/zephyr && /usr/bin/cmake -E echo
/home/mm/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(test_callback_manage.c.obj): in function `z_impl_gpio_pin_interrupt_configure':
/home/mm/zephyrproject/zephyr/include/zephyr/drivers/gpio.h:648: undefined reference to `__device_dts_ord_12'
/home/mm/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(test_callback_manage.c.obj): in function `z_impl_gpio_port_clear_bits_raw':
/home/mm/zephyrproject/zephyr/include/zephyr/drivers/gpio.h:1088: undefined reference to `__device_dts_ord_12'
/home/mm/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(test_callback_manage.c.obj): in function `gpio_add_callback':
/home/mm/zephyrproject/zephyr/include/zephyr/drivers/gpio.h:1418: undefined reference to `__device_dts_ord_12'
/home/mm/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(test_callback_manage.c.obj): in function `test_callback_enable_disable':
/home/mm/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api/src/test_callback_manage.c:224: undefined reference to `__device_dts_ord_12'
/home/mm/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(test_callback_manage.c.obj): in function `test_callback_self_remove':
/home/mm/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api/src/test_callback_manage.c:173: undefined reference to `__device_dts_ord_12'
/home/mm/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(test_callback_manage.c.obj):/home/mm/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api/src/test_callback_manage.c:121: more undefined references to `__device_dts_ord_12' follow
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/mm/zephyrproject/zephyr/build

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.15.0
 - Commit SHA:  issue can be reproduced after 975787fb5e5070add816eee7e7d46e986061ac29

rerickson1, 2022-09-13:
The BL5340 DVK uses a GPIO port expander via I2C. Before the mentioned commit above, I2C would be enabled by the boards defconfig and also enable the GPIO expander. Now it seems that https://github.com/zephyrproject-rtos/zephyr/commit/975787fb5e5070add816eee7e7d46e986061ac29 broke enabling I2C by default. I don't know enough about that previous syntax. @galak How do you think this should be fixed?

manuargue, 2022-09-19:
It also fails for board `bl5340_dvk_cpuapp_ns`. Currently blocking https://github.com/zephyrproject-rtos/zephyr/pull/49161

PR #50376 (2022-09-19): boards: bl5340_dvk: fix gpio config
If GPIO is enabled, enable I2C by default because
the BL5340 DVK uses an I2C GPIO expander.

Fixes #50344 
Fixes #50096
closed 2022-09-19 by fabiobaltieri