zephyrproject-rtos/zephyr #50106: ram_report stopped working with zephyr-sdk 0.15

jeremybettis, 2022-09-09:
**Describe the bug**
Since Zephyr requires 0.15 of the SDK now, I upgraded to that version, and then the ram report broke because of an outdated pyelftools, so I updated that to 0.29.

**To Reproduce**
For ChromeOS EC, we have a publicly visible CI system running on gitlab.com. You can see this failure at https://gitlab.com/zephyr-ec/ec/-/pipelines/635841472 but here is the excerpt:

```
ninja -C ${b} ram_report >"${BUILD_DIR}/${PROJECT}/output/${bdir}_ram_report.txt"

Traceback (most recent call last):
  File "/builds/zephyr-ec/ec/zephyr/main/scripts/footprint/size_report", line 803, in <module>
    main()
  File "/builds/zephyr-ec/ec/zephyr/main/scripts/footprint/size_report", line 779, in main
    do_simple_name_matching(elf, symbol_dict, processed)
  File "/builds/zephyr-ec/ec/zephyr/main/scripts/footprint/size_report", line 317, in do_simple_name_matching
    symbol = match_symbol_address(symlist, die,
  File "/builds/zephyr-ec/ec/zephyr/main/scripts/footprint/size_report", line 114, in match_symbol_address
    low, high = get_die_mapped_address(die, parser, dwarfinfo)
  File "/builds/zephyr-ec/ec/zephyr/main/scripts/footprint/size_report", line 84, in get_die_mapped_address
    loc = parser.parse_from_attribute(loc_attr, die.cu['version'])
  File "/home/uboot/.local/lib/python3.8/site-packages/elftools/dwarf/locationlists.py", line 307, in parse_from_attribute
    return self.location_lists.get_location_list_at_offset(
  File "/home/uboot/.local/lib/python3.8/site-packages/elftools/dwarf/locationlists.py", line 48, in get_location_list_at_offset
    raise DWARFError("For this binary, \"die\" needs to be provided")
elftools.common.exceptions.DWARFError: For this binary, "die" needs to be provided
```

**Impact**
Just an annoyance.

**Environment (please complete the following information):**
 - OS: Linux docker
 - Toolchain: Zephyr sdk 0.15
 - Commit SHA: Anything after 30af51fc5e0ccdc2b612f9912726a7211c35f13e

henrikbrixandersen, 2022-09-09:
I have no issues running `make ram_report` here using SDK 0.15.0.

jeremybettis, 2022-09-09:
> I have no issues running `make ram_report` here using SDK 0.15.0.

Hmm. I am using GCC 12 on ARM (Nuvoton npcx9m3f and npcx7), and actually not having failures on RiscV, so it might only be ARM platforms that have problems.

henrikbrixandersen, 2022-09-09:
I just confirmed a working `ram_report` with `samples/hello_world` for board `npcx7m6fb_evb`.

Have you tried upgrading the Python packages referred in `scripts/requirements.txt`?

jeremybettis, 2022-09-09:
Yes, I am getting DWARF v5. Let me see what we did that is preventing `-gdwarf-4` from working. Thank you, that was exactly the pointer I needed.

jeremybettis, 2022-09-09:
I ran these commands

```
find ec_dir/build/herobrine/build-rw -name '*.obj' | while read file; do echo $file ; /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump --dwarf=info $file | grep 'Version:.*5'; done
find ec_dir/build/herobrine/build-rw -name '*.elf' | while read file; do echo $file ; /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-readelf --debug-dump=info $file | grep 'Version: *5'; done
```

And the only files that have version 5 symbols is zephyr_pre0.elf, zephyr_pre1.elf, zephyr.elf

So I ran `ninja -v | grep zephyr_pre0` to find the link command for that binary, and added -gdwarf-4 to it, and those symbols went away.

Now to find the right place in cmake for that.

PR #50117 (2022-09-13): cmake: Add -gdwarf-4 at link time for gcc
Add a linker option for gcc to force dwarf symbols to be version 4.

Fixes #50106
Related to #35707
commit 878edeb6 (2022-09-13): cmake: Add -gdwarf-4 at link time for gcc

Add a linker option for gcc to force dwarf symbols to be version 4.

Fixes #50106
closed 2022-09-13 by mmahadevan108