zephyrproject-rtos/zephyr #50118: Twister: `--coverage-formats` Does not work despite `--coverage` added

aaronemassey, 2022-09-09:
**Describe the bug**
Invoking twister using the `--coverage-formats` with `--coverage` gives the error:

```
--coverage-formats can only be used when coverage
                        tool is set to gcovr
```

An exact command to reproduce:

```
twister -v -i --clobber -p native_posix -T tests/drivers/sensor/sbs_gauge/ -i --coverage --coverage-formats html

```

chrta, 2022-09-12:
Per default lcov and not gcovr is used by twister to create the coverage reports. You have to specify that gcovr should be used instead of lcov by adding ` --coverage-tool gcovr`
```
twister -v -i --clobber -p native_posix -T tests/drivers/sensor/sbs_gauge/ -i --coverage --coverage-tool gcovr --coverage-formats html
```

finikorg, 2022-09-14:
BTW, is coverage working now? I get report with no hits at all.

aaronemassey, 2022-09-14:
@chrta 

Ahh, the error message should probably make that a bit more clear. Thanks!

aaronemassey, 2022-09-19:
> BTW, is coverage working now? I get report with no hits at all.

Coverage definitely works for me if I just used the `--coverage` argument.

closed 2022-09-19 by aaronemassey