zephyrproject-rtos/zephyr #50119: non-IPI path of SMP is broken

justinyhuang, 2022-09-09:
**Describe the bug**
When SMP is enabled without `CONFIG_SCHED_IPI_SUPPORTED`, the idle thread could simply busy wait and yield periodically. This path is broken, which will lead to an assert failure during `do_swap`.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? 
   RISCV with SMP enabled but without IPI support.

 - What have you tried to diagnose or workaround this issue? 
  I am able to reproduce what I see with mainstream Zephyr, and below is what I do:
  • checkout HEAD of main (I was using dddb5dd6b0de9a8b8621b9ac695fd3fd980a33d6)
  • disable SCHED_IPI_SUPPORTED and other dependencies in the build 
  • Run "west build -b qemu_riscv64_smp tests/kernel/smp && west build -t run"
  • Should see the ASSERT. 

- Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
  No, but according to Andy Ross: " I'm going to place my bet that it was at or soon after this commit, which removed the last in-tree usage of k_yield() within the idle thread about a year ago: https://github.com/zephyrproject-rtos/zephyr/commit/851d14afc8941313a6f3faeb74f84ed73a33429a"

**To Reproduce**
see the steps above

**Expected behavior**
the smp tests should pass without error

**Impact**
this defect makes it impossible to bring up SMP without IPI support.

**Environment (please complete the following information):**
 - RISCV in QEMU
 
**Additional context**
A discussion thread in the [user mailing list](https://lists.zephyrproject.org/g/users/message/3075)

justinyhuang, 2022-09-09:
It looks like I cannot directly assign this issue to you, @andyross

andyross, 2022-09-13:
See linked PR, should fix it for you.  But please also try to move your new platform work onto current main and not 2.7! :)

commit 0ca7150f (2022-09-19): kernel/idle: Fix !SCHED_IPI_SUPPORTED

The requirement for k_yield() to handle "yielding" in the idle thread
was removed a while back, but it missed a spot where we'd try to yield
in the fallback loop on bringup platforms that lack an IPI.  This now
crashes, because yield now unconditionally tries to reschedule the
current thread, which doesn't work for idle threads that don't live in
the run queue.

Just make it a busy loop calling swap(), even simpler.

Fixes #50119
closed 2022-09-19 by carlescufi