zephyrproject-rtos/zephyr #50135: cannot boot up on custom board

billionfish, 2022-09-11:
**Describe the bug**
the samples/helloworld Demo can run and print well on BOARD_NRF52840DK_NRF52840. 

but this hex  file cannot boot up on  my board.

for simplest case, my custom board only use 4 pins (SWC SWD GND VCC) from [E73](https://www.aliexpress.com/wholesale?SearchText=E73%202G4M08S1C)

someone suggests set BOARD_ENABLE_DCDC=n  but not work for me.

so i need some help
thx

IvanVnucec, 2022-09-12:
Can you get console output?

billionfish, 2022-09-12:
> Can you get console output?

no output and other  information.

today I try to link the PIN to the LED but  it not lighted.

I guess the board even not boot..

nordicjm, 2022-09-12:
Probably because you haven't connected a 32KHz crystal. This isn't a zephyr problem.

billionfish, 2022-09-12:
> Probably because you haven't connected a 32KHz crystal. This isn't a zephyr problem.

[E73](https://www.aliexpress.com/wholesale?SearchText=E73%202G4M08S1C) has crystal.   And I can run other firmware on my board.