zephyrproject-rtos/zephyr #50140: ARP handling causes dropped packets when multiple outgoing packets are queued

thedjnK, 2022-09-11:
**Describe the bug**
The current ARP system in the zephyr networking stack has a very odd choice of operation, it will send an outgoing ARP request if an entry is not present, however it does not support assigning multiple packets to a handler, so in the case that a device sends multiple packets at once, e.g. 5, there will be 5 ARP requests sent out, when the first response is received, the queued first packet is sent out, the remaining 4 packets are completely ignored and eventually dropped.

**To Reproduce**
Use a networking application which sends multiple network packets in a go to a device not in the ARP table

**Expected behavior**
For one ARP request to be sent and when the response is received, for ALL pending packets that it applies to, to be sent out

**Impact**
Showstopper. I am actually amazed that there are products using the zephyr networking stack.

**Environment (please complete the following information):**
 - OS: Linux
 - Commit SHA or Version used: f1bafda5e594cc13a4efa1fbc44057104f89f8b6

rlubos, 2022-09-14:
Indeed, dropping packets when ARP transaction is already pending seems quite unfortunate... I think this could be fairly easily fixed by replacing the `struct net_pkt *` with `struct k_fifo` within [struct arp_entry ](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/l2/ethernet/arp.h#L58) - that way all pending packets could be sent out once ARP response is received.

As for the mutliple ARP requests though, is it really that a problem that mutliple request packets are being sent? Such a behaviour seems to be added into Zephyr explicitly for a reason (https://github.com/zephyrproject-rtos/zephyr/commit/2d575de085b8fcd0fcdfbb4c7c46d13705e9dfd7). There seems to be no retransmission mechanism implemented so if the orignal ARP request is lost, all of the pending packets would be discarded on timeout. Optionally, as an alternative to re-sending on consecutive packets we could implement timer-based retransmission mechanism. @jukkar What do you think?

jukkar, 2022-09-14:
I agree that we should implement a proper retransmission mechanism. The original code shows its quick and dirty past, it has served us decently but some rework is probably in place now.

thedjnK, 2022-09-14:
> As for the mutliple ARP requests though, is it really that a problem that mutliple request packets are being sent? 

It's not a problem, just a waste of packets and buffers. The thinking I had at the time was that with 3 devices on a network constantly pinging a zephyr device, that could be sufficient to prevent legitimiate communication with a host because of the default 2 ARP entry limit.

PR #50280 (2022-09-21): net: l2: ethernet: arp: Replace pending packet pointer with fifo
This fixes a bug with ARP and multiple outgoing packets with an IP that needed to be resolved, causing the first packet to go out and all others to be dropped after the timeout by having a FIFO of pending packets instead of a single packet.

Fixes #50140

I wonder if the fifo accesses should be semaphore protected e.g. to prevent an edge case of a received ARP being processed and an outgoing packet being queued at the same time?