zephyrproject-rtos/zephyr #50146: tests: kernel: mem_protect fails on nucleo_g071 with new ztest API

FRASTM, 2022-09-12:
**Describe the bug**
the test tests/kernel/mem_protect/userspace/  fails when executed on the nucleo_g071rb  and nucleo_l073rz platforms

Please also mention any information which could help others to understand
the problem you're facing:
 After the PR https://github.com/zephyrproject-rtos/zephyr/pull/49264/  and commit 8ea5cea4a7e71abc23804a400d468516dc82c2b9

maybe a question of cortex M0, 
Increasing the CONFIG_MAIN_STACK_SIZE does not change anything

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_g071rb tests/kernel/mem_protect/userspace/
3. west flash
4. See error

**Expected behavior**
test PASSED

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.1.0-3943-g8ea5cea4a7e7  ***                     
Running TESTSUITE userspace                                                           
===================================================================                   
START - test_access_after_revoke                                                      
E: thread 0x200003d0 (1) does not have permission on k_sem 0x20004dd8                 
E: permission bitmap                                                                  
E: 04 00                   |..                                                        
E: syscall z_vrfy_k_sem_take failed check: access denied                              
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000                         
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000                         
E:  xpsr:  0x00000000                                                                 
E: Faulting instruction address (r15/pc): 0xb9cc188a                                  
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0                                     
E: Current thread: 0x200003d0 (test_access_after_revoke)                              
Caught system error -- reason 3                                                       
System error was expected                                                             
 PASS - test_access_after_revoke in 0.050 seconds                                     
===================================================================                   
START - test_bad_syscall                                                              
E: Bad system call id 2147483647 invoked                                              
E: ***** HARD FAULT *****                                                             
E: r0/a1:  0x200044b8  r1/a2:  0x00000000  r2/a3:  0x200044d8                         
E: r3/a4:  0x200044d8 r12/ip:  0x00000000 r14/lr:  0x0800c9f9                         
E:  xpsr:  0x61000000                                                                 
E: Faulting instruction address (r15/pc): 0x0800c9f8                                  
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0                                   
E: Current thread: 0x200003d0 (test_bad_syscall)                                      
Caught system error -- reason 0                                                       
Wrong fault reason, expecting 3    
``` 
**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2022-09-12:
seems  similar to what happend in https://github.com/zephyrproject-rtos/zephyr/issues/49953,
 @enjiamai could you please give any indication to help fixing this issue ?

enjiamai, 2022-09-13:
Hi, @FRASTM , I did not have these boards, so could you please help to try these for debugging? Thank you very much!

(1) put this test to run firstly to see if it is failed due to previous state. 
```
diff --git a/tests/kernel/mem_protect/userspace/src/main.c b/tests/kernel/mem_protect/userspace/src/main.c
index 6364263223..b5aa587260 100644
--- a/tests/kernel/mem_protect/userspace/src/main.c
+++ b/tests/kernel/mem_protect/userspace/src/main.c
@@ -832,7 +832,7 @@ ZTEST_USER(userspace, test_unimplemented_syscall)
  *
  * @ingroup kernel_memprotect_tests
  */
-ZTEST_USER(userspace, test_bad_syscall)
+ZTEST_USER(userspace, test_1_bad_syscall)
 {
        set_fault(K_ERR_KERNEL_OOPS);
```
(2) If we skip this test, could all other tests pass? 

```
diff --git a/tests/kernel/mem_protect/userspace/src/main.c b/tests/kernel/mem_protect/userspace/src/main.c
index 6364263223..f168e2f2e0 100644
--- a/tests/kernel/mem_protect/userspace/src/main.c
+++ b/tests/kernel/mem_protect/userspace/src/main.c
@@ -832,7 +832,8 @@ ZTEST_USER(userspace, test_unimplemented_syscall)
  *
  * @ingroup kernel_memprotect_tests
  */
-ZTEST_USER(userspace, test_bad_syscall)
+//ZTEST_USER(userspace, test_bad_syscall)
+void a(void)
 {
        set_fault(K_ERR_KERNEL_OOPS);

```

By the way, these two boards both support USERSPACE, right?

FRASTM, 2022-09-13:
1. yes, both have CONFIG_USERSPACE 1
2. if executed first, the same error occurs "Wrong fault reason, expecting 3"
3. if bypassed, test completes successfully

enjiamai, 2022-09-13:
Thanks! @FRASTM , could you please give me one more favor to try this?
```
diff --git a/tests/kernel/mem_protect/userspace/src/main.c b/tests/kernel/mem_protect/userspace/src/main.c
index 6364263223..b5aa587260 100644
--- a/tests/kernel/mem_protect/userspace/src/main.c
+++ b/tests/kernel/mem_protect/userspace/src/main.c
@@ -832,7 +832,7 @@ ZTEST_USER(userspace, test_unimplemented_syscall)
  *
  * @ingroup kernel_memprotect_tests
  */
-ZTEST_USER(userspace, test_bad_syscall)
+ZTEST_USER(userspace, test_z_bad_syscall)
 {
        set_fault(K_ERR_KERNEL_OOPS);
```
I wanna make sure if this is related to certain test should be run first. (Basically it should not), thank you!

FRASTM, 2022-09-13:
in case the test_bad_syscall  is the last executed sequence, with ZTEST_USER(userspace, test_z_bad_syscall), then all  previous are PASSed and that last still block on the same error

FRASTM, 2022-09-13:
Note that the system falls into `arch_system_halt` from `handler_bad_syscall`

enjiamai, 2022-09-17:
Hi @FRASTM , I found this issue is due to the ssf parameter did not pass correctly to bad syscall handler ("kernel/userspace.c" : 915). It is null because it is not being push to stack in a right order in cortex-m0 (ARMv6) during the syscall dispatching. Actually the ssf was not passing correctly before this new ztest migration. It happened to not being "null" so this issue did not be exposed before.

Try to provide a fix #50333 for it. I verified it on nucleo_g071rb. Could you please help on verifying in other cortex-m0 boards if you have one? Thank you!

stephanosio, 2022-09-19:
Setting medium priority since this is a rather serious bug in the ARM arch port.

PR #50333 (2022-09-19): arch: arm: userspace: fix the incorrect ssf under bad syscall
The parameter ssf of the handler_bad_syscall got null pointer due to that
the R1 does not push into the stack in a right order on cortex-M0. Adjust
the pushing order of stack to make the ssf being passed correctly.

Fixes #50146.
commit d9206aa2 (2022-09-19): arch: arm: userspace: fix the incorrect ssf under bad syscall

The parameter ssf of the handler_bad_syscall got null pointer
due to that the R1 does not push into the stack in a right
order on cortex-M0. Adjust the pushing order of stack to make
the ssf being passed correctly.

Fixes #50146.
closed 2022-09-19 by carlescufi