zephyrproject-rtos/zephyr #50152: Fix internal temperature value incorrect

TDhaouST, 2022-09-12:
**Describe the bug**
 Fix the internal temperature sensor value (Die temp) incorrect when reading internal  temperature from ADC. 

This ticket is to fix the incorrect temperature sensor (Die temp) value in the ADC device-tree node for the stm32 series below:

- stm32U5
- stm32L1
- stm32L4
- stm32F3
- stm32WL

Acquired values seems inconsistent and not compatible. Example for the WL, here is the required configuration of the ADC and the die temp nodes:
		adc1: adc@40012400 {
			compatible = "st,stm32-adc";
			reg = <0x40012400 0x400>;
			clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00000200>;
			interrupts = <18 0>;
			status = "disabled";
			vref-mv = <3300>;
			#io-channel-cells = <1>;
			has-temp-channel;
			has-vref-channel;
		};

	      die_temp: dietemp {
		compatible = "st,stm32-temp-cal";
		ts-cal1-addr = <0x1FFF75A8>;
		ts-cal2-addr = <0x1FFF75C8>;
		ts-cal1-temp = <30>;
		ts-cal2-temp = <130>;
		ts-cal-vrefanalog = <3300>;
		io-channels = <&adc1 12>;
		status = "disabled";
	       };
On running sample we get: 

*** Booting Zephyr OS build zephyr-v3.1.0-4551-g411079cf1d88  ***
STM32 Die temperature sensor test
Current temperature: -171.1 �°C
Current temperature: -171.1 �°C
Current temperature: -170.4 �°C
Current temperature: -171.1 �°C

**To Reproduce**
1. west build -p auto -b nucleo_wl55jc samples/sensor/stm32_temp_sensor/
2. west flash
3. see error

**Logs and console output**
see description

**Environment (please complete the following information):**
 - OS: linux
 - Toolchain Toolchain Zephry SDK 0.15.0
 - latest main

erwango, 2022-09-14:
Removing STM32WL as this is specific. See #49897

FRASTM, 2022-09-15:
Regarding the stm32F3 disco.
The mcu is stm32F303VCT6 which refman is [RM0316 Rev 8.0](https://www.st.com/resource/en/reference_manual/rm0316-stm32f303xbcde-stm32f303x68-stm32f328x8-stm32f358xc-stm32f398xe-advanced-armbased-mcus-stmicroelectronics.pdf)  and product spec is [DS9118 Rev 14](https://www.st.com/resource/en/datasheet/stm32f303vc.pdf)

closed 2022-09-21 by fabiobaltieri