zephyrproject-rtos/zephyr #50165: boards: riscv: ite: No flash and RAM stats are shown whenever building ITE board

albertofloyd, 2022-09-12:
**Describe the bug**
No flash and RAM stats are shown whenever building ITE board

it8xxx2_evb

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/helloworld
2. rm -rf build && west build -c -b it8xxx2_evb
3.  No flash and RAM stats are shown.

**Expected behavior**
Flash and RAM stats should be shown to support footprint assessments

**Impact**
This prevent RAM and code footprint assessments, planning and optimizations.

**Logs and console output**
-- west build: building application
[1/150] Preparing syscall dependency handling
[3/150] Generating include/generated/version.h
-- Zephyr version: 3.2.0-rc1 (/home/jamezaar/ecwork/zephyr_albertofloyd), build: zephyr-v3.1.0-5147-g7c50ab1f0364
[140/150] Linking C executable zephyr/zephyr_pre0.elf
[144/150] Linking C executable zephyr/zephyr_pre1.elf
[150/150] Linking C executable zephyr/zephyr.elf

**Environment (please complete the following information):**
Linux

albertofloyd, 2022-09-12:
@itebj @RuibinChang could you help with this, looks like some flash and ram device properties are missing for the board

stephanosio, 2022-09-12:
[This pla](https://github.com/zephyrproject-rtos/zephyr/blob/7c50ab1f0364084fba228d45937ee491e312b421/boards/riscv/it8xxx2_evb/it8xxx2_evb_defconfig#L19)

itebj, 2022-09-13:
@albertofloyd 
@stephanosio 

Sorry that I re-push it and the original one has been closed.
Beg your pardon to review it again.
Thanks again for your patience and guidance.
Best regards,
BJ

PR #50170 (2022-09-14): boards: riscv: ite: enable the CONFIG_OUTPUT_PRINT_MEMORY_USAGE
Removed the CONFIG_OUTPUT_PRINT_MEMORY_USAGE in it8xxx2_evb_defconfig

fixes #50165
commit 20b78817 (2022-09-14): boards: riscv: ite: enable the CONFIG_OUTPUT_PRINT_MEMORY_USAGE

Removed the CONFIG_OUTPUT_PRINT_MEMORY_USAGE in it8xxx2_evb_defconfig

fixes #50165
closed 2022-09-14 by mmahadevan108