zephyrproject-rtos/zephyr #50188: Avoid using extra net buffer for L2 header

krish2718, 2022-09-13:
https://github.com/zephyrproject-rtos/zephyr/commit/94e89a11b7151c9f8dd8c22007e64bd30ee8b809  added change to use a dedicated net buf for the L2 header (ethernet) in this case, and as commit says it was meant to be temporary, as this unnecessarily wastes a net buffer (esp. if net buffer size is increase to 1024, we end up using only 14 bytes for Ethernet L2 header).

This should be optimized use a Linux `headroom` style concept to reserve memory to manipulate headers. A packet should only use a single net buffer to avoid memcpy when we read the packet in the L2 driver.

krish2718, 2022-09-13:
@rlubos FYI.

rlubos, 2022-09-13:
@jukkar The commit message states that "this is the first step towards getting rid of that ll reserve concept everywhere.". I see the commit is pretty old, but do you recall perhaps what was the reasoning behind such an approach?

jukkar, 2022-09-13:
Before that commit we tried to have Ethernet header in front of the L3 headers in the same net_buf. This is a bit problematic as we would need to calculate how many bytes the L2 header needs, and that is difficult to know in all the cases and then reallocate / reshuffle the buffers etc (see that commit how it re-allocates the buffers). Easiest solution was to allocate a new net_buf just for L2 header and place that as a first net_buf in the net_pkt frag chain. Of course this wastes lot of memory if you have 1k net_buf's. In this case one could use variable size data buffers by enabling `CONFIG_NET_BUF_VARIABLE_DATA_SIZE` option. This way the net_buf allocator would allocate the data part of net_buf using just enough space that the data requires.

krish2718, 2022-10-10:
I tried using `CONFIG_NET_BUF_VARIABLE_DATA_SIZE` but it doesn't event build as `CONFIG_NET_BUF_DATA_SIZE` is used in file `tcp.c`, `net_pkt.c`..

krish2718, 2022-11-08:
I tried to get the build going by hardcoding the `CONFIG_NET_BUF_DATA_SIZE` but still the packets are being dropped at L2 Ethernet at https://github1s.com/zephyrproject-rtos/zephyr/blob/HEAD/include/zephyr/net/buf.h#L1269-L1270 called from https://github1s.com/zephyrproject-rtos/zephyr/blob/HEAD/subsys/net/l2/ethernet/ethernet.c#L512 . Shouldn't we be using variable net_buf here instead of fixed?

rlubos, 2022-11-09:
> I tried to get the build going by hardcoding the `CONFIG_NET_BUF_DATA_SIZE` but still the packets are being dropped at L2 Ethernet at [...]

I've been investigating the issue recently and I have similar initial conclusions, the Ethernet layer appears to be using `net_pkt_get_frag()` for header fragment allocation, which internally relies on fixed buffers, which obviously seems wrong for the variable buffer length configuration. I don't have a fix yet, need to investigate further.

krish2718, 2022-12-01:
@jukkar @rlubos thanks for fixing the variable length net buf configuration, but we still need to discuss about using a separate net buf for L2, as this forces the drivers to make a memcpy causing performance degradation, we need a linux skb headroom style implementation.

rlubos, 2022-12-02:
Putting the case of how to allocate space for L2 header aside for a moment, I think that the driver always needs to be prepared to receive fragmented data, having the buffer large enough to fit network MTU will not always be the case. It might make sense to declare large buffers for throughput testing, but imagine a scenario where Wi-Fi runs along some 802.15.4 protocol. The networking net_buf pool is global for the system, so it doesn't make sense to set large fragment size as 802.15.4 protocol would waste ~90% of the buffer space. In this case, we need to tradeoff performance over memory usage, and the driver should be prepared for such a scenario.

krish2718, 2022-12-02:
Agreed, in the specific case we can sacrifice the performance over compatibility, but for a standalone Wi-Fi case we should aim for zero-copy (which is what we get in Linux), esp. on a low end CPU it takes about 20-40us for ~1K buffer. It's easier to check for number of fragments and then decide whether to do memcpy or not to support both cases, but we should be aiming for zero-copy in the primary usecase.

krish2718, 2024-04-12:
This feature is now being worked on in https://github.com/zephyrproject-rtos/zephyr/pull/71400

krish2718, 2025-01-13:
Verified using branch [zephyr: nrf70_zero_copy](https://github.com/krish2718/zephyr/tree/nrf70_zero_copy)  and was able to see ~29.7k RAM savings with the same peak throughput of UDP-TX 23.5Mbps using nRF7002DK.

```
== default
UDP-TX: 23.5 Mbps

Memory region         Used Size  Region Size  %age Used
           FLASH:      831200 B         1 MB     79.27%
             RAM:      454892 B       448 KB     99.16%
        IDT_LIST:          0 GB        32 KB      0.00%
        
uart:~$ net mem
Fragment length 1100 bytes
Network buffer pools:
Address         Total   Name
0x20001020      28      RX
0x20001058      27      TX
0x2000127c      28      RX DATA
0x200012a8      54      TX DATA

== CONFIG_NET_L2_ETHERNET_RESERVE_HEADER=y
UDP-TX: 23.5 Mbps

Memory region         Used Size  Region Size  %age Used
           FLASH:      831200 B         1 MB     79.27%
             RAM:      424436 B       448 KB     92.52%
        IDT_LIST:          0 GB        32 KB      0.00%
        
uart:~$ net mem
Fragment length 1100 bytes
Network buffer pools:
Address         Total   Name
0x20001020      28      RX
0x20001058      27      TX
0x2000127c      28      RX DATA
0x200012a8      27      TX DATA
```

closed 2025-01-13 by krish2718