zephyrproject-rtos/zephyr #50192: nrf_qspi_nor driver might crash if power management is enabled

onkila, 2022-09-13:
It seems that zephyr might enter to suspend state when nrf_qspi_nor-driver has locked qpsi by calling `qspi_lock()`. I did not investigate this deeper but I suspect that this might happen during some `k_msleep`-calls? Entering to suspend state is not problematic but problem is that this will cause crash as there is also `qspi_lock()` call in handling of `PM_DEVICE_ACTION_SUSPEND`. Execution will stop to that call as semaphore will stay always as taken.  nrf_qspi_nor driver shall not enter to suspend state if it is already busy.

anangl, 2022-09-19:
@onkila I don't quite see a possible crash here. If `PM_DEVICE_ACTION_SUSPEND` is requested but the driver is locked because it is still waiting for an operation to finish, the `qspi_nor_pm_action` routine will just wait in the call to `qspi_lock()` until the lock is released, i.e. until that operation finishes and the driver calls `qspi_unlock()`. What am I missing?

onkila, 2022-09-19:
First I have to say that I am not really well familiar with zephyr scheduler & power manager, but if I am correct suspend call happens from scheduler and context change won't happen until power management actions are done. So if this part of code gets stuck to something (like `qspi_lock()` call) it won't get released as context change will not happen. There was some assert that prevented this kind of behaviour and crash happened because of this. I can check the assert when I will have time to do that as I didn't write it down.

carlescufi, 2022-09-21:
@onkila it seems here that it is not clear whether this is indeed an issue with the driver. I suggest you do additional investigation and, if you conclude that this is a problem on Zephyr's side, please reopen this issue.

closed 2022-09-21 by carlescufi

onkila, 2022-10-06:
This seems to be an issue on zephyr side:

when CONFIG_PM_DEVICE is enabled and system enters suspend `pm_system_suspend()` is called from idle-thread. This calls also drivers power management function and that tries to get the semaphore. Unfortunately that causes an assert because it seems that we can not wait to get the semaphore if current thead is idle thread. There is an assert for this in sched.c line 186: 
  __ASSERT_NO_MSG(!z_is_idle_thread_object(thread));

Ozone callstack is following:
Level,Function,Stack Frame,Source,PC,Return Address,Stack Used
0,z_priq_dumb_add, 0 @ 0x20029790,sched.c:186:2,0x00058562,<inline>: 0x000585A0,232
0,add_to_waitq_locked,16 @ 0x20029790,sched.c:719:3,0x0005852C,[0x2002979C]: 0x00058600,232
0,pend,32 @ 0x200297A0,sched.c:739:3,0x000585FC,[0x200297BC]: 0x0005866C,216
0,z_pend_curr,16 @ 0x200297C0,sched.c:814:2,0x00058668,[0x200297CC]: 0x00056CB0,184
0,z_impl_k_sem_take,32 @ 0x200297D0,sem.c:143:2,0x00056CAC,[0x200297EC]: 0x00042D22,168
0,qspi_lock, 0 @ 0x200297F0,C:/Projects/kona_windows/kona/kona/build/zephyr/include/generated/syscalls/kernel.h:1035:2,0x00042D1E,<inline>: 0x00042D22,136
0,qspi_device_init,16 @ 0x200297F0,nrf_qspi_nor.c:356:2,0x00042D10,[0x200297FC]: 0x00042E00,136
0,qspi_nor_pm_action,32 @ 0x20029800,nrf_qspi_nor.c:1315:3,0x00042DFC,[0x2002981C]: 0x00075B72,120
0,pm_device_action_run,24 @ 0x20029820,C:/Projects/kona_windows/kona/zephyr/subsys/pm/device.c:80:2,0x00075B70,[0x20029834]: 0x0003019E, 88
0,pm_suspend_devices,32 @ 0x20029838,pm.c:75:3,0x0003019A,[0x20029854]: 0x00030408, 64
0,pm_system_suspend,16 @ 0x20029858,pm.c:238:3,0x00030404,[0x20029864]: 0x0005637E, 32
0,idle, 8 @ 0x20029868,idle.c,0x0005637A,[0x2002986C]: 0x00073C30, 16
0,z_thread_entry, 8 @ 0x20029870,thread_entry.c:36:2,0x00073C2E,[0x20029874]: 0xFFFFFFFF,  8
0,Top of stack - return address out of range: 0xFFFFFFFF,,,,,