zephyrproject-rtos/zephyr #50193: Impossible to connect with a peripheral with BLE and zephyr 2.7.99, BT_HCI_ERR_UNKNOWN_CONN_ID error

rubenwat, 2022-09-13:
I am having a connectivity problem with a central Bluetooth device (nrf52832) that needs to be connected to a peripheral (nrf52840), previously this connectivity worked correctly with zephyr 2.5.0 but now that I wanted to update to zephyr 2.7.99 it has stopped working correctly since I rarely get connected to the peripheral, most of the time I don't get it and I get an error=2 from the callback of connected bt_conn_cb, I also tried to increment the timeout of bt_conn_le_create_param  with no positive result. The only modification I did was the CONFIGS

This is what I have now with zephyr 2.7.99:
CONFIG_BT=y
CONFIG_BT_L2CAP_TX_MTU=252
CONFIG_BT_BUF_ACL_RX_SIZE=248 
CONFIG_BT_BUF_EVT_RX_SIZE=255
CONFIG_BT_BUF_CMD_TX_SIZE=255
CONFIG_BT_DEVICE_NAME_DYNAMIC=y
CONFIG_BT_BAS=y

and this is what I had with zephyr 2.5.0:
CONFIG_BT_L2CAP_TX_MTU=252
CONFIG_BT_L2CAP_RX_MTU=252
CONFIG_BT_RX_BUF_LEN=260
CONFIG_BT_DEVICE_NAME_DYNAMIC=y
CONFIG_BT_BAS=y

mmahadevan108, 2022-09-13:
Can you reproduce this issue on the latest main

rubenwat, 2022-09-14:
Hi, thanks for your reply it would be really tedious and time consuming for me to move to the latest 3.2.0-rc1 version as my firmware has a lot of dependencies right now. do you think we can try to make it work in this version?

jori-nordic, 2022-09-14:
@rubenwat can you manage to reproduce this using a (or multiple) samples in-tree? A branch with some sample code can also do the trick if you prefer.

rubenwat, 2022-09-16:
With those sample I do get to perform a connection, maybe the error is that my connection interval is so fast (7,5ms) but I do not understand why with zephyr 2.5.0 works correctly though.

carlescufi, 2022-10-20:
@rubenwat  we need additional details:

- Are you using vanilla Zephyr or Nordic NCS?
- Which exact revision (SHA) are you using?
- Can you please capture a sniffer trace of the failed connection and attach it to this issue?

closed 2023-01-03 by github-actions[bot]