zephyrproject-rtos/zephyr #50196: LSM6DSO interrupt handler not being called

tanoc, 2022-09-13:
**Describe the bug**
I'm trying to use a LSM6DSO (actually a LSM6DSOX, but it should be the same) with a custom board with a NRF52840.

Sometimes (mainly using tracing, `CONFIG_TRACING=y`) the LSM6DSO interrupt handler (for the accelerometer) stops being called. When it happens the interrupt port on the module looks like it's working in latched mode, with the pin set to high and not as pulsed as expected.

In the driver the pulsed mode is set in `lsm6dso_init_interrupt` with `lsm6dso_int_notification_set` but after that the driver reset the module again during `lsm6dso_init_chip` with `lsm6dso_reset_set`. Even after moving `lsm6dso_init_interrupt` after `lsm6dso_init_chip` (I'm not sure if there is a reason behind the original order) the device is still not working in pulsed mode. I think (I could be wrong) the correct way to set it in pulse mode is to set it with `lsm6dso_data_ready_mode_set(ctx, LSM6DSO_DRDY_PULSED)`. Trying with `lsm6dso_data_ready_mode_set` the issues seems solved.

**Expected behavior**
The interrupt handle should be continuously called (given it should be initialized in pulsed mode)

**Environment (please complete the following information):**
 - OS: Linux
 - Commit zephyr 3.1 branch - 2cd7fa9cbf6d8dd38745d6d08b11831e4f40ecc7

anangl, 2022-09-19:
I cannot take care of this issue as I don't know this sensor and I don't even have such one to check anything. Perhaps someone from ST could take a look. @erwango

avisconti, 2022-11-03:
Concerning the pulsed mode there is same issue as in #51488. I will provide a patch for that (and create a new bug maybe). Concerning the other part I'm not sure, I will try to reproduce it.

xyzzy42, 2022-12-23:
I also independently discovered the interrupt init order bug and have a fix.

For #51944, what is the use case of selecting pulsed or latched mode? 

In latched mode, if none of the user trigger callback handlers request accelerometer or gyro data, the interrupt will not be cleared.  This is entirely reasonable, maybe only temperature or sensor hub data is needed.  Because interrupts are edge triggered, the driver will stall with no more interrupts generated.  If interrupts were level triggered, then it would cause an interrupt loop as it would immediately re-trigger when the lsm6dso_handle_interrupt re-enables the interrupt.  So either way, latched mode breaks unless the user callback handlers request accel or gyro data.

Pulsed mode does not have this problem.  It is not necessary for user trigger callbacks to read accel or gyro data.  The interrupt could re-trigger if the data is not read before the next interrupt, but this is not a problem.  Firstly, the irq is disabled until the data is fully read, so another interrupt pulse will not trigger an irq.  Secondly, even if another irq was triggered, it would only re-schedule the handler to run again, which is not wrong or broken.

So I do not see what reason there is to support latched mode.

Also, since the interrupt is edge triggered, I don't think it is necessary to disable the irq in the irq handler and then re-enable it at the end of the work function.

avisconti, 2023-01-02:
> I also independently discovered the interrupt init order bug and have a fix.
>
> For #51944, what is the use case of selecting pulsed or latched mode?
>
> In latched mode, if none of the user trigger [...]

To be honest I'm not sure when it is better to use on or the other. I guess it may depend by the h/w setup and the way the sensor is connected to the micro, what micro is used and so on. But if you see the driver is providing a DT configuration to set pulsed or latched mode:                                                                                
```                                                                                
    drdy-pulsed:
      type: boolean
      description: |
        Selects the pulsed mode for data-ready interrupt when enabled,
        and the latched mode when disabled.
```                                                                                  
So, you may select whatever is more convenient for you.

xyzzy42, 2023-01-02:
I'm aware that both can be selected.  But, read all of my previous comment.  **Latched mode doesn't work**.  The driver isn't written properly to work in latched mode.  It will hang if an interrupt is not finished being handled before the next interrupt assertion.  Pulsed mode also doesn't work correctly.  There is a missing register bit that is not set and the driver disables and re-enables the GPIO unnecessarily.

I see two reasonable ways to fix this:

1. Fix the driver by adding more code paths to get both latched and pulsed mode to work correctly.  Since pulsed/latched is per instance, both code paths need to be compiled into the driver so it can act differently at run time for different instances.
2. Get rid of either pulsed or latched mode support.  This makes the driver smaller and simpler and improves run time speed a bit too.  Only one design needs to be considered so it's easier to avoid bugs.   Consider how current lsm6dso driver trries to do both, but gets neither correct.

I think pulsed mode makes more sense to support.  Latched mode doesn't seem to offer any improvement with this driver and with the Zephyr sensor API.  @sfeutrier used latched mode, but not on Zephyr, instead with a custom driver on the older nRF SDK.

avisconti, 2023-01-03:
> I'm aware that both can be selected. But, read all of my previous comment. **Latched mode doesn't work**. The driver isn't written properly to work in latched mode. It will hang if an interrupt is [...]

If you already have a fix (it seems so) can you prepare a PR? I will then review it.

xyzzy42, 2023-01-04:
> 
> If you already have a fix (it seems so) can you prepare a PR? I will then review it.

There is already a PR #53323.

Other bugs also keep interrupts from working correctly, fixed in that PR.

I didn't want to remove latched mode if there was some reason I hadn't considered that required it.  Motion threshold interrupts might be a reason, the lsm6dso driver does not yet support those, but it could.

closed 2023-03-21 by github-actions[bot]