zephyrproject-rtos/zephyr #50201: target_link_directories missing in Zephyr CMake extensions

SkryptX, 2022-09-13:
The CMake extension functions are missing a wrapper for [target_link_directories](https://cmake.org/cmake/help/latest/command/target_link_directories.html). This can be useful if there are very specific linker requirements. 

This would look something like this in `cmake/modules/extensions.cmake`:
```
# https://cmake.org/cmake/help/latest/command/target_link_directories.html
function(zephyr_link_directories)
  target_link_directories(zephyr_interface INTERFACE ${ARGV})
endfunction()
```

stephanosio, 2022-09-14:
cc @tejlmand

tejlmand, 2022-09-14:
> The CMake extension functions are missing a wrapper for [target_link_directories](https://cmake.org/cmake/help/latest/command/target_link_directories.html).

Just for information, then Zephyr CMake is not wrapping every CMake function just for the purpose of wrapping it.

CMake functions are wrapped when it serves the purpose of facilitating Zephyr development.
As example, `zephyr_library()` and `zephyr_library_named()` wraps CMake targets because libraries created with those functions must be whole-archived, and it sets `ZEPHYR_CURRENT_LIBRARY` so that users don't need to keep track of libname on subsequent calls to other `zephyr_library_x()` calls.
It facilitates development, especially newcomers to CMake, by ensuring the general correct `PUBLIC|PRIVATE|INTERFACE` are used.

But we don't wrap everything, so for non-standard Zephyr CMake cases, like PUBLIC / INTERFACE for source files, PUBLIC / INTERFACE on a Zephyr library include directories, then regular CMake is supposed to be used.

As there is not a general Zephyr use-case inside the Zephyr tree for `zephyr_link_directories()` then I don't see a reason for wrapping this CMake call.
It's perfectly fine to use regular CMake in a case like this.

SkryptX, 2022-09-14:
My use case is something like this and involves the currently underdeveloped Posix layer. The current `native_posix` board links to SDL2, but doesn't ensure that it is correctly linked against the library which was actually found with `pkg-config`, when SDL2 is more than just a random system library _somewhere_.

```
find_package(PkgConfig REQUIRED)
pkg_search_module(SDL2 REQUIRED sdl2)
zephyr_include_directories(${SDL2_INCLUDE_DIRS})

# missing line
target_link_directories(zephyr_interface INTERFACE ${SDL2_LIBRARY_DIRS})

zephyr_link_libraries(${SDL2_LIBRARIES})
zephyr_compile_options(${SDL2_CFLAGS_OTHER}) 
```

Therefore I would argue, that `zephyr_link_directories` is indeed missing, even though it would be currently unused in the Zephyr tree.