zephyrproject-rtos/zephyr #50205: Verify builds are reproducible in the CI

keith-zephyr, 2022-09-13:
## Introduction

Zephyr builds should be reproducible.  A checkout of Zephyr from the same commit, built with the same toolchain, should generate an identical image binary.

### Problem description

This has been proposed before (https://github.com/zephyrproject-rtos/zephyr/pull/11523 and https://github.com/zephyrproject-rtos/zephyr/pull/14593). But there are no tests that verify reproducible build in the Zephyr tree at the moment.

Furthermore, reproducible builds were broken for an unknown amount of time, but fixed with https://github.com/zephyrproject-rtos/zephyr/pull/48195.

### Proposed change

Add a new github workflow that verifies builds are reproducible. This workflow will be run on every PR.

The workflow can follow the blueprint of the Footprint Delta workflow. The new workflow would build TBD platforms, back to back, verifying the resulting binaries are identical.

Note that the build command `west build -b native_posix tests/drivers/build_all/sensor` has been known to catch problems with devicetree generation that results in non-reproducible builds.

### Dependencies

The new github workflow will block new PRs if the reproducible build test fails.

### Concerns and Unresolved Questions

Running this check against every PR will incur additional computing time and resources.

## Alternatives

Run the reproducible build check less frequently, such as nightly.  However, this will require a significant bisect effort to identify the culprit PR when any failures are detected.  The incremental cost of some additional builds on each PR seems worth the trouble.

stephanosio, 2022-09-14:
cc @marc-hb

stephanosio, 2022-09-14:
> This workflow will be run on every PR.

I do not expect this to be something that breaks often. Bi-weekly build should be fine.

gmarull, 2022-09-14:
We have many non-locked Python dependencies that are used somehow during the build process, they should be considered.

marc-hb, 2022-09-14:
Here's a list of 20+ old reproducibility fixes:
- #14593

In the same place there's an (obsolete) test script. The approach was crude but very effective:
- build once
- change as many things as possible
- build again
- diffoscope

marc-hb, 2022-09-14:
> I do not expect this to be something that breaks often. 

Agreed. Reproducibility testing and fixing is rare, but reproducibility regressions are very rare too.

> Bi-weekly build should be fine.

On the other hand, IF it's cheap and quick to run then why not run it every PR?

keith-zephyr, 2022-09-16:
Because of the amount of generated code, I'm in favor of checking on every PR.   Maybe the github workflow can be setup to run on any changes to the ./scripts directory, but also setup as a weekly run to catch problems with the actual source code.

marc-hb, 2022-11-04:
These 2 additional lines are IMHO a big step forward, please help review:
- https://github.com/zephyrproject-rtos/zephyr/pull/51954

marc-hb, 2023-04-11:
Github Actions for the Zephyr+SOF project have been routinely and successfully comparing binaries built on Linux versus Windows for a few months now:
- https://github.com/thesofproject/sof/commit/35dda2ae27c73f4be23b
- https://github.com/thesofproject/sof/actions/workflows/daily-tests.yml

To achieve this I overrode the default config change in #51954 in an SOF-specific way: https://github.com/thesofproject/sof/commit/945adb8d1660ed4

This does not catch everything (e.g.: `__DATE__`) but it indirectly provides reproducibility coverage for a lot of the Zephyr project.

Note a build is no more "reproducible" than a project is "bug-free"; fixing reproducibility bugs is a continuous activity exactly like fixing other bugs. Typically, building some code is reproducible in some Kconfiguration but fails when that Kconfiguration is changed - exactly like other bugs. Most recent example with CONFIG_ASSERT:
- https://github.com/thesofproject/sof/pull/6530
- https://github.com/thesofproject/sof/pull/7428

Switching to an old toolchain can also be very problematic:
- https://github.com/thesofproject/sof/issues/7114