zephyrproject-rtos/zephyr #50218: rcar_h3ulcb: can: failed to run RTR test cases

aaillet, 2022-09-14:
**Describe the bug**
While running `tests/drivers/can/api` test cases on Renesas R-Car platforms, the test suite fail on RTR related tests added with #47903.

The CAN controller that is equipping these boards is not able to support standard ID frames, extended ID frames & remote frames at the same time.
When the driver has been written, we then chosen to support std id and ext id frames but not remote frames.
With the new test cases, CAN API test suite is now failing on send_receive_rtr, test_send_receive_std_id_rtr and test_send_receive_ext_id_rtr when ran on R-Car boards.

RTR related test cases should maybe be disabled for R-Car boards.

**To Reproduce**
1. west build -b rcar_h3ulcb_cr7 tests/drivers/can/api/
2. (terminal n°2) picocom -b 115200 /dev/ttyUSB0
3. west flash
4. (terminal n°2) See timeout

or

1. ./scripts/twister -p rcar_h3ulcb_cr7 --device-testing -T tests/drivers/can/api/ --device-serial /dev/ttyUSB0 -vv -i
2. See error

**Expected behavior**
Test case should pass as expected.

**Impact**
The test case drivers.can.api failing, complete twister test suite is then also concerned and is reporting errors.

**Logs and console output**
```bash
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.1.0-5073-g7dc414e0cb9a  ***
DEBUG   - DEVICE: Running TESTSUITE can_api
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_add_filter
DEBUG   - DEVICE: PASS - test_add_filter in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_filters_added_while_stopped
DEBUG   - DEVICE: PASS - test_filters_added_while_stopped in 0.037 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_filters_preserved_through_bitrate_change
DEBUG   - DEVICE: PASS - test_filters_preserved_through_bitrate_change in 0.076 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_filters_preserved_through_mode_change
DEBUG   - DEVICE: PASS - test_filters_preserved_through_mode_change in 0.035 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_get_capabilities
DEBUG   - DEVICE: PASS - test_get_capabilities in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_get_core_clock
DEBUG   - DEVICE: PASS - test_get_core_clock in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_get_state
DEBUG   - DEVICE: PASS - test_get_state in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_max_ext_filters
DEBUG   - DEVICE: PASS - test_max_ext_filters in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_max_std_filters
DEBUG   - DEVICE: PASS - test_max_std_filters in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_receive_timeout
DEBUG   - DEVICE: PASS - test_receive_timeout in 0.103 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_recover
DEBUG   - DEVICE: PASS - test_recover in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_recover_while_stopped
DEBUG   - DEVICE: PASS - test_recover_while_stopped in 0.037 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_send_and_forget
DEBUG   - DEVICE: PASS - test_send_and_forget in 0.002 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_send_callback
DEBUG   - DEVICE: PASS - test_send_callback in 0.002 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_send_invalid_dlc
DEBUG   - DEVICE: E: DLC of 9 exceeds maximum (8)
DEBUG   - DEVICE: PASS - test_send_invalid_dlc in 0.003 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_send_receive_ext_id
DEBUG   - DEVICE: PASS - test_send_receive_ext_id in 0.009 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_send_receive_ext_id_masked
DEBUG   - DEVICE: PASS - test_send_receive_ext_id_masked in 0.004 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_send_receive_ext_id_rtr
DEBUG   - Waiting for device rcar_h3ulcb_cr7 to become available
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/api/src/main.c:667: send_receive_rtr: (err not equal to 0)
DEBUG   - DEVICE: receive timeout
DEBUG   - DEVICE: FAIL - test_send_receive_ext_id_rtr in 0.230 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_send_receive_msgq
DEBUG   - DEVICE: PASS - test_send_receive_msgq in 0.018 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_send_receive_std_id
DEBUG   - DEVICE: PASS - test_send_receive_std_id in 0.004 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_send_receive_std_id_masked
DEBUG   - DEVICE: PASS - test_send_receive_std_id_masked in 0.004 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_send_receive_std_id_rtr
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/api/src/main.c:667: send_receive_rtr: (err not equal to 0)
DEBUG   - DEVICE: receive timeout
DEBUG   - DEVICE: FAIL - test_send_receive_std_id_rtr in 0.235 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_send_receive_wrong_id
DEBUG   - Waiting for device rcar_h3ulcb_cr7 to become available
DEBUG   - Timed out while monitoring serial output on rcar_h3ulcb_cr7
DEBUG   - run status: rcar_h3ulcb_cr7/tests/drivers/can/api/drivers.can.api error
INFO    - 4/6 rcar_h3ulcb_cr7           tests/drivers/can/api/drivers.can.api              FAILED No Console Output(Timeout) (device 64.741s)
```

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 20.04
 - Zephyr SDK 0.15.0
 - Commit SHA : anything at the top of 097cb049168137f8d2c7520363d0b43803197578 (sha from main)

henrikbrixandersen, 2022-09-14:
@aaillet If I understand correctly, this is an issue in the receive path where the R-Car CAN controller supports two RX FIFOs, which are currently configured for STD ID data frames in FIFO0, and EXT ID data frames in FIFO1?

aaillet, 2022-09-14:
> @aaillet If I understand correctly, this is an issue in the receive path where the R-Car CAN controller supports two RX FIFOs, which are currently configured for STD ID data frames in FIFO0, and EXT ID data frames in FIFO1?

Yes, exactly !

Each FIFO is configured by two registers:
- IDE to select STD or EXT ID
- RTR to select between data and remote frames

As you said, current configuration is:
- FIFO0 accepting STD ID data frames
- FIFO1 accepting EXT ID data frames

henrikbrixandersen, 2022-09-15:
@aaillet Please see #50286 for a proposed fix.

PR #50286 (2022-09-21): drivers: can: rcar: remote transmission request filters are not supported
- Some CAN controllers do not support all filter combinations (standard 11-bit ID filters, extended 29-bit filter, data frame filters, remote transmission request). Document the expected return value (-ENOTSUP) for unsupported filter types/combinations.
- The receive path for the Renesas R-Car CAN controller supports two RX FIFOs, which are currently configured for standard 11-bit CAN ID data frames in FIFO0, and extended 29-bit CAN ID data frames in FIFO1. As this leaves no room for matching remote transmission request  [... cut]