zephyrproject-rtos/zephyr #50232: gpio_shell: Not functional anymore following DT label cleanup and deprecation

erwango, 2022-09-14:
**Describe the bug**
GPIO_SHELL provides a utility to perform few specific GPIO related actions:
```
"Configure GPIO"
"Get GPIO value"
"Set GPIO"
"Blink GPIO"
```

It could be used as follows:
```
uart:~$ gpio conf GPIOD 15 "out"
Configuring GPIOD pin 15
uart:~$ gpio set GPIOD 15 1
Writing to GPIOD pin 15
uart:~$ gpio get GPIOD 15
Reading GPIOD pin 15
Value 1
```

Since recent device tree label deprecation and cleanup, this is not functional anymore:
```
uart:~$ gpio conf GPIOD 15 "out"
uart:~$ gpio set GPIOD 15 1
uart:~$ gpio get GPIOD 15
uart:~$ 
```

**To Reproduce**
```
CONFIG_SHELL=y
CONFIG_GPIO_SHELL=y
```

**Expected behavior**
gpio shell should be functional

**Impact**
gpio shell is useless

henrikbrixandersen, 2022-09-14:
You need to use the devicetree node name instead of `GPIOD` (e.g. `gpio@deadbeef`).

erwango, 2022-09-14:
Ok, let me close and I'll open a new issue as description doesn't match a enhancement.

closed 2022-09-14 by erwango