zephyrproject-rtos/zephyr #50244: GPIO manipulation from a “counter” (ie HW timer) when Bluetooth (BLE) is enabled.

ballardr, 2022-09-14:
**Describe the bug**
Hi, I’m having issues controlling GPIO from a “counter” (ie HW timer) when Bluetooth (BLE) is enabled.

My program flow is that I set up counter to interrupt after a period equivalent of 1200baud. Each time the interrupt runs, I set the bit logic on my bus. It’s basically a software UART (it's SDI-12), but I needed to implement a few extra features which Zephyr’s UART driver didn’t support (mainly sending serial breaks).

The issue is, when BLE is enabled, there is a delay before the logic switching actually occurs and then the switching occurs much faster than the baud rate so corrupts the data. See the below image. Note, the code for these captures are identical, the only difference is BLE being enabled.

[image]

The issue being seen, is indicative IMO, of a thread (system thread I assume) doing something and then suddenly being able to service a queue/semaphore. When I searched, I found this about Bluetooth hogging resources (see https://github.com/zephyrproject-rtos/zephyr/issues/20392).

Do counter callbacks occur on the HW timer’s interrupt or do the HW timer’s interrupt actually signal via a semaphore (or queue) and the callback occurs via the user or system thread?
- If it’s the later, is there any way to actually run code on the interrupt itself in a “zephyr” way (ie without actually implementing the ISRs directly)? It would seem silly if this was the case as it renders counters next to useless, you may as well just use k_sleep() as all timing benefits of using a HW timer are lost
- If it’s the former, does gpio_pin_set() set gpio directly or does it ask the system thread to do this?
- Is there some other reason why GPIOs are not being set from the ISR

The "bug" is that GPIO manipulation from a timer should occur when the timer interrupt occurs but it isn't.

**Expected behavior**
GPIO manipulation from a timer should occur when the timer interrupt occurs

**Impact**
Showstopper - it renders counter useless and causes all data on bus to be corrupted. 

**Logs and console output**
See image above

**Environment (please complete the following information):**
 - Linux (Zephyr SDK) and Windows (GCC)
 - v2.7 LTS (I can’t upgrade to Zephyr 3 or above due to https://github.com/zephyrproject-rtos/zephyr/discussions/43428)

ballardr, 2022-09-14:
I did have this as a discussion (https://github.com/zephyrproject-rtos/zephyr/discussions/50216) but after some sleep, I thought I'd add as a issue and add details about likely being a queue/semaphore servicing issue

cvinayak, 2022-09-15:
> Do counter callbacks occur on the HW timer’s interrupt or do the HW timer’s interrupt actually signal via a semaphore (or queue) and the callback occurs via the user or system thread?

Counter callbacks occur in the timer's interrupt context. User applications are to use kernel features to defer the handling to a work queue or threads to achieve lower ISR latencies.
 
> If it’s the later, is there any way to actually run code on the interrupt itself in a “zephyr” way (ie without actually implementing the ISRs directly)? It would seem silly if this was the case as [...]

The counter callbacks are ISRs, hence already providing the means to execute user code in ISRs. Kernel features can be used as necessary to defer operations.

>If it’s the former, does gpio_pin_set() set gpio directly or does it ask the system thread to do this?

GPIO are directly set. These are set of CPU instructions, themselves consume CPU cycles in doing so.

>Is there some other reason why GPIOs are not being set from the ISR

Various IRQs at priority levels will introduce ISR latencies. Even IRQs at same priority level have the natural order of priority level that will introduce latencies across ISRs.

> The "bug" is that GPIO manipulation from a timer should occur when the timer interrupt occurs but it isn't.

This is not bug, real time system need careful profiling of interrupt duty cycle, CPU use, ISR latencies introduced by IRQ priorities
and interrupt locking etc. Bluetooth Controller uses couple of IRQs that can introduce latencies (even upto 1ms). I suggest you use hardware features like peripheral-to-peripheral interconnect or similar to achieve near real-time requirements.

closed 2022-09-15 by cvinayak