zephyrproject-rtos/zephyr #50255: Test process crash when run twister with --coverage

mengxianglinx, 2022-09-15:
**Describe the bug**
Run twister with --coverage, test cases passed, test process break when dump coverage info, in handler.log:

 git bisect" it to commit 30af51fc5e

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p qemu_x86 -T tests/kernel/mutex/mutex_api/ --coverage

**Logs and console output**
 GCOV_COVERAGE_DUMP_START
 *E: Page fault at address 0x1f889f9a (error code 0x0)
 E: Linear address not present in page tables
 E: Access violation: supervisor thread not allowed to read
 E: PDE: not present
 E: EAX: 0x1f889f9a, EBX: 0x00000000, ECX: 0x00000002, EDX: 0x00000000
 E: ESI: 0x00000000, EDI: 0x001058d3, EBP: 0x0014d778, ESP: 0x0014d760
 E: EFLAGS: 0x00000246 CS: 0x0008 CR3: 0x00169000
 E: call trace:
 E: EIP: 0x0010e208
 E:      0x0010e4a4 (0x1f889f9a)
 E:      0x0011cce5 (0x0)
 E:      0x00105906 (0x0)
 E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
 E: Current thread: 0x140400 (main)
 E: Halting system

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain Zephyr SDK 0.15.0
 - Commit SHA or Version used: b27c5d73ef

fabiobaltieri, 2022-09-20:
cc @stephanosio

enjiamai, 2022-09-22:
Hi @mengxianglinx , I fixed this by #50477, could you please help review it?

PR #50477 (2022-09-22): tests: framework: fix code coverage report on zephyr-sdk-0.15
The GCC/GCOV version over 12 has slight format change of the gcno and gcda. Make some adaption in the gcov dump function to fix the code coverage report.

Mainly two places change:
1. Added the checksum in the struct gcov_info. This fix the crash in qemu_x86, and mps2_an385 when run with --coverage.

2. Adjust the GCOV_TAG_FUNCTION_LENGTH accroding to gcov-io.h. It's length unit is caculated by bytes now.

Fixes #50255.
Fixes #50257.
commit 77986382 (2022-09-22): tests: framework: fix code coverage report on zephyr-sdk-0.15

The GCC/GCOV version over 12 has slight format change of the gcno
and gcda. Make some adaption in the gcov dump function to fix the
code coverage report.

Mainly two places change:
1. Added the checksum in the struct gcov_info. This fix the crash
in qemu_x86, and mps2_an385 when run with --coverage.

2. Adjust the GCOV_TAG_FUNCTION_LENGTH accroding to gcov-io.h. It's
length unit is caculated by bytes now.

Fixes #50255.
Fixes #50257.
closed 2022-09-22 by fabiobaltieri