zephyrproject-rtos/zephyr #50256: I2C on SAMC21 sends out stop condition incorrectly

benediktibk, 2022-09-15:
Running an I2C master on a SAMC21 results in an incorrect end of the transaction. The last byte will be read and the master will send out correctly a NACK. But after this another 9 clocks follow with the stop condition after this. Actually the stop condition should come right after the NACK, without any additional clocks in between.

The issue is that the stop condition is triggered in the context of the thread, instead of during the interrupt:

> Commands can only be issued when either the Client on Bus Interrupt flag (INTFLAG.SB) or Host on Bus Interrupt
flag (INTFLAG.MB) is '1'.

Source: https://ww1.microchip.com/downloads/aemDocuments/documents/MCU32/ProductDocuments/DataSheets/SAM-C20-C21-Family-Data-Sheet-DS60001479J.pdf (Page 587).

I was able to fix it in our local fork of zephyr with the following patch:

```
diff --git a/drivers/i2c/i2c_sam0.c b/drivers/i2c/i2c_sam0.c
index d557bb4..939c94d 100644
--- a/drivers/i2c/i2c_sam0.c
+++ b/drivers/i2c/i2c_sam0.c
@@ -154,6 +154,7 @@ static void i2c_sam0_isr(const struct device *dev)
 	if (status & SERCOM_I2CM_INTFLAG_MB) {
 		if (!data->msg.size) {
 			i2c->INTENCLR.reg = SERCOM_I2CM_INTENCLR_MASK;
+			i2c->CTRLB.bit.CMD = 3;
 			k_sem_give(&data->sem);
 			return;
 		}
@@ -169,6 +170,7 @@ static void i2c_sam0_isr(const struct device *dev)
 			 * require write synchronization.
 			 */
 			i2c->CTRLB.bit.ACKACT = 1;
+			i2c->CTRLB.bit.CMD = 3;
 		}
 
 		*data->msg.buffer = i2c->DATA.reg;
@@ -492,23 +494,6 @@ static int i2c_sam0_transfer(const struct device *dev, struct i2c_msg *msgs,
 			return -EIO;
 		}
 
-		/*
-		 * Only send a stop if after this message:
-		 *   - it is explicitly requested that we send a stop, or
-		 *   - we are not conducting a restart, with more messages to follow
-		 *
-		 * Note: nothing validates the flags, so default to a stop if a stop is
-		 * requested... do not let a restart request override it
-		 */
-		bool send_stop = (data->msgs->flags & I2C_MSG_STOP)
-			|| !((data->msgs->flags & I2C_MSG_RESTART) && (data->num_msgs > 1));
-
-		if (send_stop) {
-			while (!i2c->STATUS.bit.CLKHOLD) {
-			}
-			i2c->CTRLB.bit.CMD = 3;
-		}
-
 		data->num_msgs--;
 		data->msgs++;
 	}
```

But I assume this is not the whole story. First of all, the interrupt flags are actually already cleared at this point again, but within the context of the same interrupt routine just a few lines of code earlier on. So technically not 100% correct, but it works at least in my setup. Second, I am uncertain if all the flags (data->msgs->flags) are considered correctly this way.

It took a look at the current version of the driver (b27c5d73ef09278186dc8858dfbed2a33dff6031) and I strongly assume that the same bug is still there.

nandojve, 2022-09-20:
Hi @benediktibk ,

Thank you for sharing this with us. Please, open a PR with your fix so I can add more people to look at it.
It is important mention that there is no support for SAMC21 on current Zephyr but maybe the driver uses same IP and we may have same bug anyway. Maybe people from #50384 may help to check too.

@attie-argentum This maybe could be interesting to you look. Let me know if you could reproduce, if you have the time.

benys, 2022-09-22:
@benediktibk please correct me: I2C works now with master code, but not efficient.
You fix that sercom stops clock/transmission faster?

I read L21 / D21 / C21 datasheet and they mention the same thing. 

> Commands can only be issued when either the Client on Bus Interrupt flag (INTFLAG.SB) or Host on Bus Interrupt
flag (INTFLAG.MB) is '1'.

(I tested on my board i2c "master" code with one i2c sensor and it works)

attie-argentum, 2022-10-03:
@benediktibk - thanks for your patch, I've reviewed it and would like to also relocate the STOP on error - see #50933.

I've credited you in the commit message, but if you'd prefer to make the PR I'm more than happy to pass it to you.

benediktibk, 2022-10-04:
I'm absolutely happy that you prepared the PR, I just wasn't yet able to find the time to prepare a test setup as a preparation for creating the PR. Thanks mate!

closed 2022-10-04 by fabiobaltieri

TomzBench, 2023-02-24:
I have a same54p20a part and i can see this patch is merged, but i still seem to have an extra byte clocked at the end. Does anyone have a `atsame54_xpro` board that can confirm if this is fixed there?

attie-argentum, 2023-02-24:
@TomzBench - I don't think the SAM E54 is covered by the `SAM0` drivers... have you checked for a similar issue in the relevant drivers? (e.g: [`i2c_sam_twi.c`](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/i2c/i2c_sam_twi.c) or [`i2c_sam_twihs.c`](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/i2c/i2c_sam_twihs.c))

TomzBench, 2023-02-24:
@attie-argentum hello,

I added a discussion here: https://github.com/zephyrproject-rtos/zephyr/discussions/55170

It seems that the `sam0` support is intentionally pretty minimal but im not sure if im following the driver correctly, maybe you can take a look

PR #50933 (2022-10-04): drivers: i2c: sam0: Resolve spurious trailing 0xff on reads
Previously, the I2C bus would have an extended idle period on write operations, and a spurious trailing byte on read operations (after a final NACK).

This patch relocates the issuance of the stop condition from the driving thread's context, into the ISR.

The following screenshot shows the "_before_" signalling:
[image]

The following screenshot shows the "_after_" signalling:
[image]

This PR fixes #50256.

Tested on SAMR34 (aka SAML21).