zephyrproject-rtos/zephyr #50293: sta feature on dma perpheral feature

hakehuang, 2022-09-16:
**Is your enhancement proposal related to a problem? Please describe.**
in many DMA peripheral there are some address alignment on source and target address.

**Describe the solution you'd like**
There are 2 option to resolved this

- option 1:

currently we can use the existing dma-buf-alignment define in dma dts feature, and let application to select the purpore one.

- option 2

We need define dma-buf-alignment in each perpherial dts feature. and all dma enabled peripheral will be required to do so.
and together with dma-buf-alignment in the exiting dma dts to control the src and dst address alignment

**Additional context**
refer to below PR for more details
[dma-buf-alignment
](https://github.com/zephyrproject-rtos/zephyr/pull/49793)

teburd, 2022-09-16:
I think it would be useful to have the alignment property known somehow in the pair of DMA + Peripheral, the best place for that to me and maybe others would think otherwise, would be on the peripheral, but if there's some better way to describe some attributes tied to the pair of devices that would be best.

@mbolivar-nordic or @galak any thoughts on how best to do this?

mbolivar-nordic, 2022-09-16:
@teburd I'm confused. From @hakehuang 's description it sounds like the DMA peripheral itself is where the alignment restrictions are. E.g. a memory to peripheral transfer can only read from 4-byte aligned memory addresses and things like that. When you say "pair of DMA + Peripheral", what do you mean? Are there additional alignment constraints that are specific to the combination of the DMA controller and a specific peripheral address it's talking to that you need to capture?

hakehuang, 2022-09-19:
@mbolivar-nordic , I think pair means each peripheral with DMA support need add such constrains.
for peripheral dma, the src is usually a 4 byte aliged 32 bit word. but it depends on the peripheral. so if we need add this to each peripheral, would this be better that we create a common dts for all peripheral that supports dma?

hakehuang, 2022-09-22:
> Just to be clear, when you say "peripheral DMA", you mean the DMA controller, right?

no it is the peripheral that has DMA can read from,  such register is usually 4 bytes address aligned, but it is platform dependant.

mbolivar-nordic, 2022-10-12:
I'm sorry, I'm still lost. If we're talking about a peripheral register then its address is fixed so I don't get what we need to do -- we can figure out its alignment by looking at the bits in the register.

Maybe you can point me to some real hardware with example devicetrees for option 1 and option 2 from the PR description?

If not,  I think we may need to call a meeting.

hakehuang, 2022-10-12:
> I'm sorry, I'm still lost. If we're talking about a peripheral register then its address is fixed so I don't get what we need to do -- we can figure out its alignment by looking at the bits in the register address.
> 
> Maybe you can point me to some real hardware with example devicetrees for option 1 and option 2 from the PR description?
> 
> If not, I think we may need to call a meeting.

soem platfrom may support 16/32 bits and 64 bits in the future.
and the question now is: 
we have define a dma-buf-alignment in dma engine dts, which limits the DMA buffer address alignment, but for different peripheral do we need define a common dts attribute which can be inherient and applys to different peripheral? as the two address alignment may not the same.