zephyrproject-rtos/zephyr #50294: test-ci: timer_behavior: mimxrt1170_evk_cm7/1160: test failure

hakehuang, 2022-09-16:
**Describe the bug**
Timer variance and standard deviation is below defined acceptable values

**To Reproduce**
Steps to reproduce the behavior:
cd tests/kernel/timer/timer_behavior
1. mkdir build; cd build
2. west build -b mimxrt1170_evk_cm7
3. west flash
4. See error

**Expected behavior**
test pass

**Impact**
timer accuracy

**Logs and console output**
```
*** Booting Zephyr OS build v3.2.0-rc1-18-g53af1ba59e29  ***
Running TESTSUITE timer_behavior
===================================================================
START - test_periodic_behavior
periodic timer behavior test
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 4096, expected period 1000 us, samples 10000, period 32.768000 cycles
statistics samples 10000, rollovers 0, mean 1220.684814 us, variance 1.862310 us, stddev 1.364665 us, min 1098.632812 us, max 1220.703125 us
statistics samples 10000, rollovers 0, mean 39.999400 cycles, variance 0.002000 cycles, stddev 0.044717 cycles, min 36 cycles, max 40 cycles
timer start cycle 646, end cycle 400640, total time 12206848.144531, expected time 10000000.000000, difference 2206848.144531
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/main.c:166: timer_behavior_test_periodic_behavior: (max_us <= max_us_bound is false)
Longest timer period too long (off by more than 10%)
FAIL - test_periodic_behavior in 12.287 seconds
===================================================================
TESTSUITE timer_behavior failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [timer_behavior]: pass = 0, fail = 1, skip = 0, total = 1 duration = 12.287 seconds
- FAIL - [timer_behavior.test_periodic_behavior] duration = 12.287 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 04dea6255811a9be4fde024507d7c15b
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux,)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA: v3.2.0-rc1-18-g53af1ba59e29

hakehuang, 2022-09-16:
@mmahadevan108

mmahadevan108, 2022-09-16:
@hakehuang, this seems like a duplicate of Issue https://github.com/zephyrproject-rtos/zephyr/issues/49576

closed 2022-09-16 by mmahadevan108