zephyrproject-rtos/zephyr #50306: Not able to flash stm32h735g_disco - TARGET: stm32h7x.cpu0 - Not halted

maxstreitberger, 2022-09-16:
When using the new Zephyr-SDK-0.15.0, I cannot flash the `stm32h735g_disco` board.

If I build the `zephyr/samples/hello_world project` for the `stm32h735g_disco` and try to flash it, I receive this error:
```
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/max/Desktop/zephyrproject/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.11.0+dev-00724-g42b6471c1 (2022-08-17-18:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate

Info : clock speed 1800 kHz
Info : STLINK V3J9M3 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.300602
Error: BUG: can't assert SRST
Info : [stm32h7x.cpu0] Cortex-M7 r1p2 processor detected
Info : [stm32h7x.cpu0] target has 8 breakpoints, 4 watchpoints
Error: read_memory: read at 0x5c001004 with width=32 and count=1 failed
Error executing event examine-end on target stm32h7x.cpu0:
/home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg:236: Error: read_memory: failed to read memory
in procedure 'stm32h7x_dbgmcu_mmw' called at file "/home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 169
in procedure 'stm32h7x_mmw' called at file "/home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 259
in procedure 'stm32h7x_mrw' called at file "/home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 241
at file "/home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 236
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32h7x.cpu0      hla_target little stm32h7x.cpu       reset

Error: read_memory: read at 0x5c001004 with width=32 and count=1 failed
Error executing event examine-end on target stm32h7x.cpu0:
/home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg:236: Error: read_memory: failed to read memory
in procedure 'ocd_process_reset' 
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 876
in procedure 'stm32h7x_dbgmcu_mmw' called at file "/home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 169
in procedure 'stm32h7x_mmw' called at file "/home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 259
in procedure 'stm32h7x_mrw' called at file "/home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 241
at file "/home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 236
Error: timed out while waiting for target halted
TARGET: stm32h7x.cpu0 - Not halted
FATAL ERROR: command exited with status 1: /home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/max/Desktop/zephyrproject/zephyr/boards/arm/stm32h735g_disco/support -s /home/max/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/max/Desktop/zephyrproject/zephyr/boards/arm/stm32h735g_disco/support/openocd.cfg '-c init' '-c targets' -c 'reset halt' -c 'flash write_image erase /home/max/Desktop/zephyrproject/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /home/max/Desktop/zephyrproject/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```
I compared the target file `stm32h7x.cfg` found in `~/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/` from the Zephyr-SDK-0.14.2 with the file from the Zephyr-SDK-0.15.0 and saw some changes, which I then reverted one by one until I noticed that the change in line `145` broke the flashing process. After changing the line back to `reset_config srst_only srst_nogate`, I could flash the `stm32h735g_disco` board again.

I have also noticed that after applying the change to the `stm32h7x.cfg` file, I have to flash the `nucleo_f429zi` board twice before the flash command succeeds. Afterwards, I can flash it as usual until I flash the `stm32h735g_disco` again.

**For clarification:**
| Flash order                         | Success?      |
| ---------------------------- |------------------------------|
| stm32h735g_disco    | :heavy_check_mark:   |
| nucleo_f429zi            | :x:                                |
| nucleo_f429zi            | :heavy_check_mark:   |
| nucleo_f429zi            | :heavy_check_mark:  |
| stm32h735g_disco    | :heavy_check_mark:  |
| nucleo_f429zi            | :x:                               |
| nucleo_f429zi            | :heavy_check_mark:  |

This problem only occurred after the switch to the new Zephyr-SDK version. In the previous `0.14.2` version, everything worked fine, and I can use the `stm32h735g_disco` without any issues on macOS, where I do not use the SDK. 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b stm32h735g_disco -s zephyr/samples/hello_world/ -p always
2. west flash

**Impact**
annoyance

**Environment (please complete the following information):**
 - OS: Linux
 - main branch
 - Zephyr SDK 0.15.0

erwango, 2022-09-20:
I've contacted STM32 openocd maintainer for this specific issue.

Meanwhile, if this is blocking, I suggest to use stm32cubeprogrammer runner. Reducing criticality to low since there is an alternate solution.

erwango, 2022-09-23:
@maxstreitberger I suspect an openocd configuration issue.
Would  you mind trying to replicate the configuration provided for [nucleo_h743zi](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nucleo_h743zi/support/openocd.cfg).

Regarding nuclo_f429zi, I'll open a dedicated ticket, as I've found the root cause, and I'm not sure it applies to H7

erwango, 2022-09-30:
@maxstreitberger Can you try: https://github.com/zephyrproject-rtos/zephyr/pull/50860 ?

PR #50860 (2022-09-30): boards: stm32h735g_disco: Provide a working openocd configuration
Openocd configuration for this board was broken. Fix it. Tested on both SDK 0.15.0 and SDK 0.15.1-RC1.
Debug is also functional

Fixes #50306
commit dcf4201d (2022-09-30): boards: stm32h735g_disco: Provide a working openocd configuration

Openocd configuration for this board was broken. Fix it.
Tested on both SDK 0.15.0 and SDK 0.15.1-RC1.
Debug is also functional

Fixes #50306
closed 2022-09-30 by mmahadevan108