zephyrproject-rtos/zephyr #50330: Fail to find GICv3 Redistributor base address for Cortex-R52 running in a cluster different than 0

manuargue, 2022-09-17:
**Describe the bug**
It seems that https://github.com/zephyrproject-rtos/zephyr/commit/68b10e8572d9ad4633a8c9f71e9ebf48bb8019ee has broken the support for Cortex-R52 to run in a different cluster than the 0. In Cortex-R52, `GICR_TYPER` doesn't use Aff1-3 (RAZ, https://developer.arm.com/documentation/100026/0101/Chunk723471919), so only `GICR_TYPER.Aff0` can be compared against `MPIDR.Aff0`. When running let's say in cluster 1, core 0 (`MPIDR.Aff0=0`, `MPIDR.Aff1=1`), it's not possible to find the redistributor base address with the current implementation.

**To Reproduce**
Reproducing this issue with `fvp_baser_aemv8r_aarch32` seems not possible because it supports a single cluster (cluster 0). I have tested with the board being introduced in https://github.com/zephyrproject-rtos/zephyr/pull/49161

**Expected behavior**
GICv3 initialization must be able to find the GIC Redistributor base address when running Zephyr on Cortex-R52 in any cluster/core.

**Impact**
Affects this PR currently in review https://github.com/zephyrproject-rtos/zephyr/pull/49161

**Environment (please complete the following information):**
 - OS: Linux, Windows
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 2829de75d1ae88dc53bc6e85d8f8e59ac3cca8d3

manuargue, 2022-09-17:
Hi @SgrrZhf @carlocaione, could you help me take a look on this? thanks

carlocaione, 2022-09-17:
@povergoing

povergoing, 2022-09-20:
Let me take a look

povergoing, 2022-09-20:
@manuargue Not sure if I understand your issue correctly, you meant your platform has several independent distributor regions like `#redistributor-regions` > 1 or your platform has a special redistributor-stride size?
> When running let's say in cluster 1, core 0 (MPIDR.Aff0=0, MPIDR.Aff1=1), it's not possible to find the redistributor base address with the current implementation.

Could you give me more information about why the current code cannot find the GIC Redistributor base address? In terms of https://developer.arm.com/documentation/100026/0101/generic-interrupt-controller/gic-functional-description/gic-distributor-memory-map, if your platform follows this one, and has only one region, it should work. Iterating all redistributors by matching the aff(0 ~3) is logically the right way to get the redistributor base that is how Linux does
https://github.com/torvalds/linux/blob/master/drivers/irqchip/irq-gic-v3.c#L945-L946

manuargue, 2022-09-20:
@povergoing our platform has several independent distributor regions implemented as detailed in the documentation above.

However, for Cortex-R52 the `GICR_TYPER` fields for [aff1 to aff3 are not used and read as zero](https://developer.arm.com/documentation/100026/0101/Chunk723471919):
[image]

The current algorithm to find the redistributor base address, works by comparing `GICR_TYPER.Aff0~3` against `MPIDR.Aff0~3`. For a core in cluster 1, `MPIDR.Aff1` will be set to `1` but `GICR_TYPER.Aff1` will read as `0`, making the comparison fail in all iterations and eventually being unable to find a base address.

povergoing, 2022-09-20:
Thanks, understood. I actually doubt if it's right to design in this way. But it's also reasonable since clusters are independent. Anyway, fixing it seems easy by adding a `CONFIG_ ` just using `aff0` when comparing. Do you think it's enough for your case?

manuargue, 2022-09-20:
Yes, a fix would be to compare only `Aff0` from both registers when building for Cortex-R52.

manuargue, 2022-09-21:
@povergoing just to know, should I provide a fix or you are working on it already?

povergoing, 2022-09-21:
I am working on it already, will send the PR soon

povergoing, 2022-09-21:
@manuargue would you like to take a look and check if it would fix your issue? I have verified it with tweaked FVP

manuargue, 2022-09-21:
The bugfix works on our SoC, thanks for the fix 👍

Btw it would be great if you can share the tweaked FVP parameters you used, just out of curiosity and in case is needed in the future.

povergoing, 2022-09-21:
> Btw it would be great if you can share the tweaked FVP parameters you used, just out of curiosity and in case is needed in the future.

Sure, it is `-C cluster0.cpu0.MPIDR-override=0x80000100`, overriding the original 0x80000000 with 0x80000100 and 100 means aff1= 0. BTW, you could check with all the params by running `./FVP_BaseR_AEMv8R -l`

PR #50466 (2022-09-21): Fix the failure of finding GICv3 Redistributor base address for Cortex-R52
Fix #50330
Some platforms like R52 platform[1] only use aff0 to match MPIDR.aff and GICR.aff. Introduce `GIC_V3_RDIST_MATCHING_AFF0_ONLY` to set the matching method only using aff0. With this config enabled, the matching function will find the target redistributor by comparing the aff0 only. This config is set to y if CPU_CORTEX_R52 is enabled. 

[1] https://developer.arm.com/documentation/100026/0101/Chunk723471919
closed 2022-09-21 by carlescufi