zephyrproject-rtos/zephyr #50339: UART/serial bitbang driver

jonnykl, 2022-09-18:
**Is your feature request related to a problem? Please describe.**
Hardware UART does not always allow custom baudrates. A bitbanging UART driver would provide a solution in such a case.

**Describe the solution you'd like**
There are already I2C and SPI bitbanging drivers implemented. Maybe a similar driver could also be implemented for UART?

**Describe alternatives you've considered**
\-

**Additional context**
\-

dianjixz, 2022-09-20:
And W1 bitbang drivers,haha!

dcpleung, 2024-02-09:
I2C and SPI have dedicated clock line so timing is not that critical on the bit-bang driver. However, as UART (in general) has no dedicated clock line, timing is very critical on the software side to to manipulating the signal, where even a timer interrupt coming it will disrupt I/O (let alone NMIs). So it is highly unlikely there can be a usable implementation. So closing it.

closed 2024-02-09 by dcpleung