zephyrproject-rtos/zephyr #50344: bl5340_dvk_cpuapp: undefined reference to `__device_dts_ord_14`

stephanosio, 2022-09-19:
**Describe the bug**

```
undefined reference to `__device_dts_ord_14'
```

**To Reproduce**

```
scripts/twister -v -N -M -p bl5340_dvk_cpuapp -s samples/basic/blinky/sample.basic.blinky
```

**Expected behavior**

Test builds successfully

**Impact**

Unable to build tests and samples for `bl5340_dvk_cpuapp`

**Logs and console output**

```
Loading Zephyr default modules (Zephyr base).
-- Application: /__w/zephyr/zephyr/samples/basic/blinky
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /__w/zephyr/zephyr/.cache
-- Zephyr version: 3.2.0-rc2 (/__w/zephyr/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: bl5340_dvk_cpuapp
-- Found host-tools: zephyr 0.15.0 (/opt/toolchains/zephyr-sdk-0.15.0)
-- Found toolchain: zephyr 0.15.0 (/opt/toolchains/zephyr-sdk-0.15.0)
-- Found Dtc: /opt/toolchains/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /__w/zephyr/zephyr/boards/arm/bl5340_dvk/bl5340_dvk_cpuapp.dts
-- Generated zephyr.dts: /__w/zephyr/zephyr/twister-out/bl5340_dvk_cpuapp/samples/basic/blinky/sample.basic.blinky/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /__w/zephyr/zephyr/twister-out/bl5340_dvk_cpuapp/samples/basic/blinky/sample.basic.blinky/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /__w/zephyr/zephyr/twister-out/bl5340_dvk_cpuapp/samples/basic/blinky/sample.basic.blinky/zephyr/dts.cmake
Parsing /__w/zephyr/zephyr/Kconfig
Loaded configuration '/__w/zephyr/zephyr/boards/arm/bl5340_dvk/bl5340_dvk_cpuapp_defconfig'
Merged configuration '/__w/zephyr/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to '/__w/zephyr/zephyr/twister-out/bl5340_dvk_cpuapp/samples/basic/blinky/sample.basic.blinky/zephyr/.config'
Kconfig header saved to '/__w/zephyr/zephyr/twister-out/bl5340_dvk_cpuapp/samples/basic/blinky/sample.basic.blinky/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/toolchains/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    TC_RUNID

-- Build files have been written to: /__w/zephyr/zephyr/twister-out/bl5340_dvk_cpuapp/samples/basic/blinky/sample.basic.blinky
[1/172] Preparing syscall dependency handling

[2/172] Generating misc/generated/syscalls_subdirs.trigger
[3/172] Generating include/generated/version.h
-- Zephyr version: 3.2.0-rc2 (/__w/zephyr/zephyr), build: v3.2.0-rc2
[4/172] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[5/172] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[6/172] Generating include/generated/driver-validation.h
[7/172] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h, include/generated/otype-to-size.h
[8/172] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
[9/172] Generating include/generated/offsets.h
[10/172] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc32c_sw.c.obj
[11/172] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj
[12/172] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc32_sw.c.obj
[13/172] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc8_sw.c.obj
[14/172] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc16_sw.c.obj
[... 135 lines cut ...]
[150/172] Building C object zephyr/kernel/CMakeFiles/kernel.dir/condvar.c.obj
[151/172] Building C object zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/nrf_qspi_nor.c.obj
[152/172] Building C object modules/hal_nordic/nrfx/CMakeFiles/modules__hal_nordic__nrfx.dir/__w/zephyr/modules/hal/nordic/nrfx/drivers/src/nrfx_gpiote.c.obj
[153/172] Linking C static library zephyr/drivers/flash/libdrivers__flash.a
[154/172] Linking C static library modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a
[155/172] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
[156/172] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timeout.c.obj
[157/172] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mempool.c.obj
[158/172] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mutex.c.obj
[159/172] Building C object zephyr/kernel/CMakeFiles/kernel.dir/work.c.obj
[160/172] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
[161/172] Linking C static library zephyr/kernel/libkernel.a
[162/172] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map 
: && ccache /opt/toolchains/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/__w/zephyr/zephyr/twister-out/bl5340_dvk_cpuapp/samples/basic/blinky/sample.basic.blinky/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/cmse/libarch__arm__core__aarch32__cortex_m__cmse.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -L"/opt/toolchains/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/thumb/v8-m.main/nofp"  -L/__w/zephyr/zephyr/twister-out/bl5340_dvk_cpuapp/samples/basic/blinky/sample.basic.blinky/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m33  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,--fatal-warnings && cd /__w/zephyr/zephyr/twister-out/bl5340_dvk_cpuapp/samples/basic/blinky/sample.basic.blinky/zephyr && /usr/local/bin/cmake -E echo
/opt/toolchains/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(main.c.obj): in function `main':
/__w/zephyr/zephyr/samples/basic/blinky/src/main.c:42: undefined reference to `__device_dts_ord_14'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA: 2829de75d1ae88dc53bc6e85d8f8e59ac3cca8d3

stephanosio, 2022-09-19:
cc @rerickson1 @greg-leach

manuargue, 2022-09-19:
Perhaps related to https://github.com/zephyrproject-rtos/zephyr/issues/50096?

PR #50376 (2022-09-19): boards: bl5340_dvk: fix gpio config
If GPIO is enabled, enable I2C by default because
the BL5340 DVK uses an I2C GPIO expander.

Fixes #50344 
Fixes #50096
closed 2022-09-19 by fabiobaltieri