zephyrproject-rtos/zephyr #50345: Network traffic occurs before Bluetooth NET L2 (IPSP) link setup complete

wentong-li, 2022-09-19:
**Describe the bug**
This issue is about BT network l2 implementation. When creating a L2CAP channel, after receiving the request from initiator, BT RX thread on the responder will handle the packet, and the call stack would be:

(beginning of call stack omitted)

- l2cap_recv(),
- which calls le_conn_req();
- which calls l2cap_chan_accept().

In this function, it call server's "accept" callback first, and then channel ops' "connected" callback if "accept" callback returns true. In current implementation, the "connected" callabck will execute "net_if_up(conn->iface)" operation. And along this function call, it

- calls net_if_up(conn->iface),
- which calls iface_ipv6_start(iface) as the condition !(l2_flags_get(iface) & NET_L2_POINT_TO_POINT) is true,
- which calls net_if_start_dad(iface) as the condition IS_ENABLED(CONFIG_NET_IPV6_DAD) is true,
- which calls net_if_ipv6_start_dad(iface, &ipv6->unicast[i]),
- which calls net_ipv6_start_dad(iface, ifaddr),
- which calls net_ipv6_send_ns(iface, NULL, NULL, NULL, &ifaddr->address.in6_addr, true),
- which calls net_send_data(pkt).

And this will eventually pass pkt data to BT TX thread.

After calls l2cap_chan_accept() returns, inside le_conn_req(), l2cap_send() is called to send l2cap layer response to initiator. This indeed passes some data to BT TX thread as well. At this moment, the responder will consider this l2cap channel is setup already, and the bt net_if is up. When the responder switches to BT TX thread, it will send these data to the initiator. 

The problem occurs on the initiator side. After sending the l2cap channel setup request, it is waiting for the response. However, as the responder will send network traffic data before the l2cap response. This means some unexpected data arrives before the channel is setup on initiator's side. Then it will discard the net traffic data, and consider the attempt of creating l2cap channel fails. Even if the l2cap response arrives, as the channel is discarded, the reponse is not handled. And the bt net_if is considered remaining down.
 
**To Reproduce**
Steps to reproduce the behavior:
1. use echo server and echo client samples, with CONFIG_NET_L2_BT_MGMT=y and CONFIG_NET_L2_BT_SHELL=y. On client sample, CONFIG_NET_CONFIG_BT_NODE=y, and on server sample, CONFIG_NET_CONFIG_BT_NODE=n.
2. platform: nrf5340dk_nrf5340_cpuapp
3. zephyr revision: I am using nrf connect sdk, which has cunstomed zephyr of v3.1.99-ncs1. But I think this problem also exists in original zephyr.
4. On server sample side, use shell to send "net_bt connect <peer's bt mac address> random", and the bt connection is setup. Send the command again to setup l2cap channel. The server/initiator side will show message like

> [00:00:22.607,604] <err> bt_l2cap: No credits to receive packet
> [00:00:22.706,817] <wrn> bt_l2cap: Ignoring data received while disconnecting
> [00:00:22.756,835] <wrn> bt_l2cap: Ignoring data received while disconnecting
> [00:00:22.756,958] <wrn> bt_l2cap: Ignoring data received while disconnecting
> [00:00:22.806,884] <err> bt_l2cap: Cannot find channel for ident 1
> [00:00:22.856,994] <wrn> bt_l2cap: Ignoring data received while disconnecting
> [00:00:22.906,951] <wrn> bt_l2cap: Ignoring data received while disconnecting
> [00:00:23.606,292] <wrn> bt_l2cap: Ignoring data for unknown channel ID 0x0040
> [00:00:24.606,323] <wrn> bt_l2cap: Ignoring data for unknown channel ID 0x0040

and if you run "net iface", the interface is down.
5. On client/responder side, it shows

> [00:00:25.797,729] <inf> net_config: Interface 1 (0x20001160) coming up

and if you run "net iface", the interface is up.

**Expected behavior**
The l3 activity should start only after the l2 link is fully setup. 

**Impact**
bt-based link layer connection cannot be setup. And discrepancy of the link state is observed.

**Additional context**
To fix it, currently I postpone the net_if_up(conn->iface) call by warpping it as a delayable work and just submit it to system queue with some delay (e.g. 10ms) in the connected callback. This can ensure the l2cap response is sent before any network traffic. But I think the problem should be solved in bluetooth stack. It should call "connected" callback after sending the l2cap_connection_rsp message. And this needs bluetooth expert to verify if it meets bluetooth specification, and if it breaks existing BT stack, as ATT and SM also use l2cap channel.

kruithofa, 2022-09-21:
@wentong-li Can you see if PR #50476 fixes this issue?

wentong-li, 2022-09-22:
Seems not. IMO, the receiver of L2CAP channel connection req does things in wrong order. Currently, it first call "accept" callback, then "connected" callback, and finally send L2CAP channel connection rsp. And I am sending data over this l2cap channel in "connected" callback. It is fine for the receiver, but the sender of L2CAP channel connection req is waiting for L2CAP channel connection rsp but gets other data before the response. Maybe we can reorder the sequence as call "accept" callback -> send L2CAP channel connection rsp -> call "connected" callabck. It is more intuitive that the "connected" callback is called after all setup is done, including sending L2CAP channel connection rsp.

closed 2022-12-06 by github-actions[bot]