zephyrproject-rtos/zephyr #50354: ztest_new: _zassert_base : return without post processing

hakehuang, 2022-09-19:
**Describe the bug**
the  _zassert_base  will return directly, and if we have malloc before, there is no way to handle this fault(free memory), which cause all following cases fails.

yperess, 2022-09-19:
Another option is to use a dedicated heap and `k_heap_alloc`. You can then assume that in the test suite's after function it is safe to clear the whole heap .

hakehuang, 2022-09-20:
> If you'd like the zexpect API to be added. Please make a GitHub issue linking
> this one and assign it to @yperess and myself.

the code snipper is as below
```
	/* Allocate buffers */
	tmp1 = malloc(MAX_MATRIX_DIM * MAX_MATRIX_DIM * sizeof(float64_t));
	zassert_not_null(tmp1, ASSERT_MSG_BUFFER_ALLOC_FAILED);

	tmp2 = malloc(MAX_MATRIX_DIM * MAX_MATRIX_DIM * sizeof(float64_t));
	zassert_not_null(tmp2, ASSERT_MSG_BUFFER_ALLOC_FAILED);

	output = calloc(length, sizeof(float64_t));
	zassert_not_null(output, ASSERT_MSG_BUFFER_ALLOC_FAILED);
```
so the question is whether the zexpect can deallocate the memory allocated from sys heap? IMHO @yperess proposal is more clear to me.

yperess, 2022-09-20:
> so the question is whether the zexpect can deallocate the memory allocated from sys heap? IMHO @yperess proposal is more clear to me.

It's not that the expect API would handle the `free` for you, it's that on failure, it won't bail on the test function. So for example:

```
/* Allocate buffers */
tmp1 = malloc(MAX_MATRIX_DIM * MAX_MATRIX_DIM * sizeof(float64_t));

/* Assert that malloc returned non NULL, if the assert fails, just bail. */
zassert_not_null(tmp1, ASSERT_MSG_BUFFER_ALLOC_FAILED);

/* Make a function call that writes to the allocated memory */
do_stuff(tmp1);

/* A failed expect will mark the test as failed, but will continue execution */
zexpect_equal(tmp1[0], 5);

free(tmp);
 ```

hakehuang, 2022-09-21:
> /* Assert that malloc returned non NULL, if the assert fails, just bail. */
> zassert_not_null(tmp1, ASSERT_MSG_BUFFER_ALLOC_FAILED);

the problem is  above line bails(return), and following testcase will get executed with the allocated memory not freed.

hakehuang, 2022-09-30:
> @hakehuang - `z_assert_*` and `z_expect_*` APIs don't have any handling for freeing memory allocated in a test case. Both these APIs on failure will mark the test as failed. The difference is that the `z_assert` will halt execution and any lines afterwards are not executed, which what we mean by "bail". The suggestion to use `z_expect` is that the test is still marked as failure and the test will continue to run which will have your memory clean-up logic. So something like this will clean-up any allocated memory
> 
> ```
> tmp1 = malloc(5);
> z_expect_not_null(tmp1) // PASS
> tmp2 = malloc(HUGE); // Unexpected NULL is returned
> z_expect_not_null(tmp2); // FAILS -- test is marked as failure but test continues
> 
> free(tmp1);  // Cleans-up tmp1
> free(tmp2);  // Cleans-up tmp2 although its NULL
> ```

@asemjonovs  thanks a lot, so we need change z_assert* to z_expect* for cases that need afterwards clean up. @enjiamai , maybe you can apply this change in cmsis-dsp cases

enjiamai, 2022-10-08:
Thanks, got it. @hakehuang , will give it a try soon.

closed 2022-12-22 by github-actions[bot]