zephyrproject-rtos/zephyr #50381: BLE: Connection slows down massively when connecting to a second device

CodingGhost, 2022-09-19:
I have a program which reads DMX512 over uart at 250kbaud and controls some peripherals based on the DMX values.
when 1 device is connected, everything is super fast, but as soon as I connect a second device, things slow down massively, with 1 device, I get latency of about 10-15ms, with 2 devices its over a second.

the first strange thing is, that even if I disconnect the second device, the link stays slow until I disconnect all devices and reconnect the first one. I found out that on the last device disconnecting, Im starting advertising and that seems to fix the latency somehow?!
So I tried doing a advertising start and stop immidiately after every disconnect and now it gets fast as soon as I diesconnect the second device. I have no Idea why this is happening.

the next thing I tried was locating the part where all the time is wasted, but I couldnt find it. 
I measured the time from receiving a new value until the write_without_response reports error 0 and it always stays well below a millisecond, so the time has to be wasted somewhere after that in the zephyr code I guess? does the write without response return before the data is aired? if so, the culprit would have to be between the write call and the radio transmission.

I have written other programs that have multiple connections and they do not suffer from these problems, so Im not sure if zephyr is to blame, but still all this is very strange.

cvinayak, 2022-09-20:
@CodingGhost Please retry your application with latest upstream main branch, I suspect you may have stumbled on the issue fixed here: #50126

If you are unable to use the latest main, you may try cherry-pick this commit: https://github.com/zephyrproject-rtos/zephyr/commit/eff946a9ade2facc72454de290427d020f603e22

mmahadevan108, 2022-09-27:
@CodingGhost Does that fix the issue you are seeing.

CodingGhost, 2022-09-28:
no it doesnt. Im still testing stuff and will report back when I found something new.

CodingGhost, 2022-09-29:
so I tested alot today and found out it had something todo with the settings subsystem.
I then made the effort and updated my sample app to zephyr 3.1 with NCS and suddenly it worked!
so although I dont know why the cherry pick didnt work (could be user error) updating to 3.1 solves my issues.

Ps. I already tried upgrading my complete program to 3.1 and NCS (coming from 2.7 PlatformIO) but Im struggling hard. I get tons of linker errors and VScode doesnt show the errors on the Problems tab, so I always have to read through the (not even highlighted) build output. can somebody maybe assist me here a little bit? would be appreciated.(discord is John Warren#2723)

carlescufi, 2022-10-19:
Fixed in 3.1.

closed 2022-10-19 by carlescufi