zephyrproject-rtos/zephyr #50389: Allow twister to be called directly from west

yperess, 2022-09-19:
**Is your enhancement proposal related to a problem? Please describe.**
Different projects might place `zephyr` in different places, as such, calling `twister` may use different paths. This makes it difficult for developers that work on multiple projects. Additionally, twister uses different flags from west (`-p` and `-b` being the most obvious) which just makes things even more confusing, if possible we should find a way to consolidated these two tools and ideally make twister callable from west.

**Describe the solution you'd like**
1. Add a west subcommand to be able to run twister (`west test`)
2. Make a pass at the west/twister arguments and combine as many of them as possible into the same argument parser (optional)

**Describe alternatives you've considered**
Adding west features to twister and making twister installable (this does not scale)

mbolivar-nordic, 2022-09-20:
I would want it called `west twister`. Otherwise I'm on board.

mbolivar-nordic, 2022-09-20:
@PerMac FYI. I'm guessing we could add a `west twister --v2` or `west twister-v2` at some point as well.

mbolivar-nordic, 2022-09-20:
> **Describe alternatives you've considered**
> Adding west features to twister and making twister installable (this does not scale)

@yperess are you aware of https://github.com/zephyrproject-rtos/twister ?

This effort actually does seem to have some of the alternatives you've considered so I am curious why you think it does not scale.

yperess, 2022-09-20:
> @yperess are you aware of https://github.com/zephyrproject-rtos/twister ?
>
> This effort actually does seem to have some of the alternatives you've considered so I am curious why you think it does not [...]

Twister tightly couples with ztest to parse the logs and generate the test plan. That means that if I have two projects with different versions of zephyr, I'll need two versions of twister installed.

nashif, 2022-09-21:
> @PerMac FYI. I'm guessing we could add a `west twister --v2` or `west twister-v2` at some point as well.

There shall be just one twister, if we decide to move to a new implementation, it will be done in a transparent way. I am not on board with supporting 2 different versions of twister at any given time.

nashif, 2022-09-21:
> twister uses different flags from west (`-p` and `-b` being the most obvious)

Twister (sanitycheck) was there before west and it always used -p for platform, we can make this less confusing by using the -b also for twister..

nashif, 2022-09-21:
more thoughts...

Are we looking at a plain wrapper here, i..e `west twister` to call the right twister from the current workspace or a more than that? would `west twister` accept the twister command line options as if you are calling `twister` directly or is there going to be some level of translation, i.e. `-b` becomes `-p` etc.?

I am in favor of more integration and providing more value, i.e. I would prefer `west test -b <board> <test/sample>` which would translate using twister as a backend to run testsuites defined in a test/sample app, so an extension on top of what we have now...
- `west build ...` (build only)
- `west build -t  run ...` (build and run)
- `west build test ...`

This could be further enhanced to run tests on devices if connected.

a wrapper and solving the original issue of 
> Different projects might place zephyr in different places

can be resolved in so many different ways. Twister for me for example is always in the path and pointing to the current `ZEPHYR_BASE/scripts/twister`....

asemjonovs, 2022-09-23:
As I'm learning about west, here are some thoughts I have (please let me know if I'm missing any details)

1. Since twister already builds and tests - west should use twister as a backend for both `build` and `test` commands. Doesn't make sense to me to have different behavior here.  Which means this becomes an exercise of invoking twister with `--build-only` or `--test-only` args and translating west args to compatible twister args.  
2. Going the route of `west twister` command - This will basically be a wrapper for twister and I think should use the same arguments as twister. Only job would be to resolve or override current `ZEPHYR_BASE` environment variable.
3. Sounds like the original issue is when there's multiple west workspaces, invoking the right twister thats matches the version of the workspace is causing issues. I noticed there are configurations that can be added to the west manifest. We could add the path to the correct twister script there (`west config twister.path <path_to_twister>`).  Creating a `west twister` command will look up the twister path based on this config, if not available it will resolve the path of based on current path.  This will be an additional step for the developer when setting up their environment, but anytime they invoke `west twister` in their directory structure should find this path and invoke the right version of twister. I've noticed currently many python scripts resolve `ZEPHYR_BASE` with the current location of itself when the env variable isn't set and doesn't check the west config manifest.

Thoughts?

PerMac, 2022-09-29:
I think we are reaching a bit too deep level of inception: getting a wrapper to wrap over another wrapper wrapping this wrapper :). 
I believe the hierarchy should go like this: cmake -> west -> twister.  I think I am convinced with Anas's argumentation (presented in various other places, e.g. https://github.com/zephyrproject-rtos/zephyr/pull/36729#pullrequestreview-752684699) that the core build system scripting shouldn't be part of twister, since twister is a test runner and users should be able to do general building/flashing without it. IMO the fact that twister is doing it's own scripting for building is against this axiom but I think it is a legacy issue, since as Anas pointed out, twister existed before west. In twisterV2 we do building by directly calling west, but will yet to see if this can be the general solution for all configurations. 
@asemjonovs having this in mind I don't agree with point 1.) of your thoughts. I think we should aim at the opposite direction: unifying the workflows by getting twister to base on west.

As for 2.), 3.) and @yperess  's comment about scalability: Wouldn't a virtual environments solve this issue? I found working without them quit tedious if one works with more projects. Different versions of zephyr often have different requirements for various tools. I think working with virtual envs based on requirements files is a nice way of managing dependencies. And since twisterV2 was mentioned here, IMO having twister as a tool with it's own versioning and not be based on the code checkout with a given zephyr version has a lot of benefits https://discord.com/channels/720317445772017664/733037738068148335/976818388057862154. With separate virtual envs I think you shouldn't have issues with managing multiple versions of twister. Also note, that this is already how we deploy west. It is a package on it's own, and not a part of the source code. Also it already happens that for different zephyr versions you need different minimal west version.

yperess, 2022-09-29:
Adding @keith-zephyr and @jackrosenthal to this discussion since we just started an internal discussion about a virtual environment internally

nashif, 2023-01-30:
this is merged, closing.

closed 2023-01-30 by nashif